davidrapan/ha-solarman #441: SmartLoadOff vs SmartLoadOn input type

rromrrom, 2025-03-04:
### Description

Thanks for very good integraion. Good job!

I'm using deye inverted and what I see that **smartloadOff** is a slider input, while **smartloadOn** is numeric input field.
I would expect they are both of the same type and should use same input type.
I would prefer input field, not slider. But it is up to developer.

Thanks.

[image]

### Reproduction Steps

basic integratino setup, no specific steps.
Then open settings->devices and services -> solarman -> click on configred device to see all entities and inputs.

### Log

this is UI issue.

### Version

25.02.14

### Home Assistant Core Version

2025.2.5

### Home Assistant Operating System Version

-

davidrapan, 2025-03-04:
Hi @rromrrom, thanks!

Here is the [fix: SmartLoad Off input mode for Deye](https://github.com/davidrapan/ha-solarman/commit/de81fc07533ea57f1e29bdf3a4e7f6edc83a74c3)

rromrrom, 2025-03-04:
@davidrapan thanks for super sonic fix!

closed 2025-03-10 by davidrapan