davidrapan/ha-solarman #446: Function code 313 instead of 3

qbasar, 2025-03-06:
### Description

Hey @davidrapan,

I'm not sure if I should turn to pysolarmanv5 project in this case, but since I'm using this integration, I'll try here first.

I have a Bluesun BSE12KH3 inverter paired with a Solarman's LSW5-BLE stick logger. I have been trying to wrap my head around this whole process for a couple of days now, starting from a complete newb standpoint.

Following your guide/understanding to some extent how pysolarmanv5 works, I tried to follow the steps using the afore_hybrid.yaml as recommended. The local IP is correct (have no problem accessing it through web) as well as the serial number.

No matter whan I did, I was always unable to setup any devices/entities, the integration just keeps trying to reload with a "function code 131 instead of 3".

[image]

### Reproduction Steps

What I then did is try the register scan with a following example for "PV1 Voltage" register and this was the result:

When decoding the received frames (using pysolarmanv5's decoder from utils), I mostly get a failed CRC and the data always comes out to be the same. (0500 or 018302c0f1).

Occassionally I manage to get function code 131 decoded as well, but it didn't happen in this two examples.

If I understand correctly, should "Data" be the requested value in this case?

### Log

```
Scanning input registers
DEBUG:pysolarmanv5.pysolarmanv5:[353xxxxxxx] SENT: a5 17 00 10 45 8f 00 xx xx xx d2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 04 02 2b 00 01 40 7a b1 15
DEBUG:pysolarmanv5.pysolarmanv5:[353xxxxxxx] RECD: a5 10 00 10 15 8f 76 xx xx xx d2 02 01 a1 5e 5b 32 82 1a 00 00 5f 3d 6e 35 05 00 70 15
Finished scanning input registers
Scanning holding registers
DEBUG:pysolarmanv5.pysolarmanv5:[353xxxxxxx] SENT: a5 17 00 10 45 90 00 xx xx xx d2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 2b 00 01 f5 ba a6 15
DEBUG:pysolarmanv5.pysolarmanv5:[353xxxxxxx] RECD: a5 13 00 10 15 90 77 xx xx xx d2 02 01 a1 5e 5b 32 82 1a 00 00 5f 3d 6e 35 01 83 02 c0 f1 a7 15
Finished scanning holding registers
```

### Version

25.02.14

### Home Assistant Core Version

2025.3.0

### Home Assistant Operating System Version

14.2

fuzzy25, 2025-03-06:
fwiw, I was trying to setup solarman and ran into the same issue, using a DLS-W logger and a Solis RHI-5K-48ES-5G inverter .. following...

davidrapan, 2025-03-07:
Hi, you need to select appropriete profile. **Auto** is only for Deye compatible inverters...

qbasar, 2025-03-07:
Any ideas what I should try in my case?

Since Bluesun BSE15KH3 should be supported by the afore_hybrid profile, my BSE12KH3 should be in the same boat, right? Difference in power shouldn't change anything here, I believe?

However I'm guessing the registers could either be different or I'm unable to read the correct data from the responses.

davidrapan, 2025-03-07:
You didn't provide any logs so I have no idea...

qbasar, 2025-03-07:
Cheers, I'll solve it myself.

fuzzy25, 2025-03-07:
@qbasar RHI-5K-48ES-5G is my inverter, I tried with several templates but none of them work, sometimes different error code .. I was hoping the type would fall under solis-hybrid but that didn't work either. Should it work with any of the templates? I found in the hybrid template the following comment.. _Solis RHI-1P3~6K-48ES-5G | 3~6kW | Single Phase | Hybrid Inverter_, but mine doesn't have the "1P3~6" in its type number.

davidrapan, 2025-03-07:
Simply try every profile there is. 😉

fuzzy25, 2025-03-08:
hi, I don't know what made the difference but it now just works with the hybrid yaml file 🤔 😄

uswong, 2025-03-09:
I am having Sofar hybrid inverter `Sofar ESI-5K-S1` with logger s/n 27xxxxxxxx version no. `LSW3_15_MQTT_270A_1.10`, when I installed Stephan's version of Solarman, it installed successfully with a device and more than one hundred entities, but all values shown are `Unknown` except the `Sofar Status_connection` shown as `disconnected`. 

So, I installed David's version of Solarman and I used the `sofar_lsw3.yaml` but it also didn't work with `failed setup` error same as the one mentioned by @qbasar above.  As advised by @davidrapan, I tried many other profiles, and quickly I got it working like a charm with `sofar_g3hyb.yaml`.

Many thanks for the work done by Stephan and David.

davidrapan, 2025-03-09:
Your device is compatible w/ `afore_hybrid` so that's what you should use. If there are any issues you need to figure out which registers are causing them using debug logging.

qbasar, 2025-03-09:
[home-assistant_solarman_2025-03-09T15-32-29.476Z.log](https://github.com/user-attachments/files/19150517/home-assistant_solarman_2025-03-09T15-32-29.476Z.log)

This is what keeps happening with `afore_hybrid`, or any config in that matter.

davidrapan, 2025-03-09:
Try to remove entities w/ registers from `0x00CE` to `0x00D3` and run it w/o them.

qbasar, 2025-03-09:
I removed the mentioned entites, but the same issue kept happening, except with the different addresses.

I only left one single address in the `afore_hybrid`, that being `0x09C4`, which _should be_ "Running status", but the result is still the same.

It seems the inverter must have completely different addresses/registers. I'm trying to find some additional information documentation, but I haven't seen much of these inverters around.

[home-assistant_solarman_2025-03-09T16-46-13.534Z.log](https://github.com/user-attachments/files/19150905/home-assistant_solarman_2025-03-09T16-46-13.534Z.log)

davidrapan, 2025-03-09:
> It seems the inverter must have completely different addresses/registers.

I don’t believe that because I know someone w/ your device who’s running integration w/ `afore_hybrid`. Try inverter firmware upgrade then.

qbasar, 2025-03-09:
Alright, thanks. I can try that.

In the meantime, I was trying to scan larger numbers of addresses and managed to receive some values at odd registers, decimals 40050-40060. Strange.

Nothing from 0-2000.

qbasar, 2025-03-09:
Well, update didn't help as there didn't seem to be any, however I have managed to find correct register ranges.

I have found values from decimal 30000-35000 and some at 40000-40100. I have tried one of the values and it now successfully integrates. No idea what the value is yet, but that shouldn't be an issues sorting it out.

[image]

All I can do now is map them out, once I have enough time on my hands, and share the .yaml/documentation files here in case this happens to anyone else. 🤷‍♂ 

Perhaps it is worth nothing the inverter is brand new (received in December 2024) and they might have changed some things. I have found some data in my inverter stating it's a type: "EPH12KHL", which points to inverters such as this one: https://www.thinkpower.com.cn/hybrid-three-phase4-12kw.html

Which also visually looks the same.

Anyway, sorry for the extensive rambles and thanks for nudging me in the right direction.

I'll come back once this is functional as it should be.

davidrapan, 2025-03-10:
Try to contact your manufacturer, maybe they'll help you out. 😉

davidrapan, 2025-03-14:
Hi @neofral, can you tell us your opinion on this?

qbasar, 2025-03-17:
Just wanted to update that I'm still waiting for a response from a manufacturer, contacted two different sources.

They don't seem to be eager to share this information for some reason.

neofral, 2025-03-24:
> Hi [@neofral](https://github.com/neofral), can you tell us your opinion on this?

sry didn't go into github for quite some time. 

[image]
i have this setting on advanced. ip address of the home assistant and the port
this is the file i'm using:

[bluesun (2).txt](https://github.com/user-attachments/files/19412214/bluesun.2.txt)
use StudioCodeServer and browse to: config/custom_components/solarman/inverter_definitions

create a file with the name bluesun.yaml and paste the content of the txt file attached here.
i'll keep my eyes open to help you

sven0833, 2025-03-30:
> sry didn't go into github for quite some time.
>
> [image] i have this setting on advanced. ip address of the home assistant and the port this is the file i'm using:
>
> [bluesun [...]

hello i got a bluesun 10kva inverter i tried this...but i cant seem to make it work on home assistant

qbasar, 2025-03-30:
Hey, doesn't work for me as well.

Also, I still haven't received any replies from Bluesun. Asked Solarman if they can help and pointed me toward the manufacturer.

At this point I'm hoping some more data of this inverter pops up online so I can set it up.

sven0833, 2025-03-30:
> Hey, doesn't work for me as well.
> 
> Also, I still haven't received any replies from Bluesun. Asked Solarman if they can help and pointed me toward the manufacturer.
> 
> At this point I'm hoping some more data of this inverter pops up online so I can set it up.

oh so ill just wait if theres a result in the future then... im not that tech savy on thing ang just always follow tutorials on youtube.

davidrapan, 2025-05-04:
Keep Alive

davidrapan, 2025-05-08:
Hi @qbasar, @neofral and @sven0833, try it with [megarevo_r-3h.yaml](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/megarevo_r-3h.yaml)

qbasar, 2025-06-02:
Hi @davidrapan,

tried the mentioned config, doesn't work to me.

[image]

davidrapan, 2025-06-02:
Which registers are failing?

qbasar, 2025-06-12:
This is the result, it doesn't seem to be calling the correct registers:

[home-assistant_solarman_2025-06-12T11-03-17.361Z.log](https://github.com/user-attachments/files/20707660/home-assistant_solarman_2025-06-12T11-03-17.361Z.log)

david9600, 2025-06-13:
From that log, the inverter seems to be rejecting address x'16A2' (dec 5794), and this is repeated a few times.

The Modbus slave rejects a command by setting the first bit in the fn code byte of its response (128+3 = 131).  In the hex packet dump is "01 83 02", where 01 is the slave address, 83 is an exception response to fn code '03', and the reason code in the next byte is '02', meaning the requested storage address is not valid for the slave. 

I had the same issue a few months ago with a Sofar inverter until I found the right registers for my model....

qbasar, 2025-06-13:
Yeah, I've been scanning the registers on and off for some time now and trying to map them out, but some are almost impossible to identify. I've spotted values such as L1, L2, L3, frequencies of those, which are pretty standard numbers, and other similar values. The ones that can switch pretty regularly are harder to pinpoint, especially when comparing to the cloud values, since there's always some kind of a delay.

As mentioned before, I've manage to get responses from addresses ranging from decimal 30000-35000 and some at 40000-40100.

If I'm not mistaken, I didn't see any similar addresses with other inverters.

david9600, 2025-06-13:
OK... that output from the scanner shows it's sending the actual address across the link (x'7531' = decimal 30001), and getting a positive response from the inverter - so it's not a software offset.
I think I'd be chasing the inverter installer and the manufacturer again to see if they'll provide the register details.

qbasar, 2025-06-17:
Okay, so I've made some progress. I've tried with the manufacturer once more, no response.

In the meantime, I've been trying to pick out as many registers as I can and map them out. This is the current result:

[Bluesun_BSE12KH3_register_map.xlsx](https://github.com/user-attachments/files/20773178/Bluesun_BSE12KH3_register_map.xlsx)

There are still unidentified registers in there, however, I am now able to implement it into Home Assistant as I wanted it to.

[image]

[image]

Here is the inverter definition file, based on the afore_hybrid.yaml:

[bluesun_BSE12KH3.zip](https://github.com/user-attachments/files/20773312/bluesun_BSE12KH3.zip)

If anyone in need of this files manages to identify any other registers, please feel free to do so. 
I am also happy to help with any similar issues anyone may encounter and hopefully a solution can be found.

david9600, 2025-06-17:
Definitely looks like you have control of it now ... nice work!

qbasar, 2025-06-17:
Thank you for your help!