davidrapan/ha-solarman #447: Error during change storage control mode

vymak, 2025-03-06:
### Description

When I want to change the "**Storage control mode**" for example from "**Self use**" to "**Passive**", it displays a message, see screen.

[image]

Profil: **sofar_g3hyd.yaml**
Logger: **LSE-3**

[image]

### Reproduction Steps

Change "**Storage control mode**" from "**Self use**" to "**Passive**"

### Log

[home-assistant_solarman_2025-03-06T18-02-27.033Z.log](https://github.com/user-attachments/files/19113239/home-assistant_solarman_2025-03-06T18-02-27.033Z.log)

### Version

25.02.14

### Home Assistant Core Version

2025.3.0

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-07:
Hi, can you try to write using Developer tools >> ACTIONS >> Solarman: Write Single Register (Modbus Function Code 6):
* Address: 4368
* Value: 0

vymak, 2025-03-07:
Hi @davidrapan, I tried and here is the result

[image]

davidrapan, 2025-03-10:
And w/ Solarman: Write Multiple Registers (Modbus Function Code 16) it shows that **unpack** error?

hoschilo, 2025-03-12:
I have same issue with HYD 20 KTL (Sofar) and yes Function Code 16 shows unpack error (requires a butter of 4 bytes)

pedroza26, 2025-03-13:
I have the same issue on:

Inverter mode: **SOFAR HYD6KTL-3PL**
Logger: **LSE-3**
Solarman ver: **25.03.12_1**

Whatever I am trying to change from configuration tab on the device it's giving the same kind of error like below

[image]

[image]

Wizwalli, 2025-03-14:
> And w/ Solarman: Write Multiple Registers (Modbus Function Code 16) it shows that **unpack** error?

Hey @davidrapan yes also on the write multiple registers
See debug logs below

2025-03-14 10:24:51.462 DEBUG (MainThread) [custom_components.solarman.services] write_multiple_registers: <ServiceCall solarman.write_multiple_registers (c:01JP9ZH046ATAK7PQV1KN0WMYD): device=xxxx, address=4488, values=[65374, 65535, 65374, 65535]>
2025-03-14 10:24:51.463 DEBUG (MainThread) [custom_components.solarman.device] [2108379728] Scheduling request
2025-03-14 10:24:51.463 DEBUG (MainThread) [custom_components.solarman.device] [2108379728] Call 16 ❘ 0x10 ~ 4488 ❘ 0x1188: {'registers': [65374, 65535, 65374, 65535]} ...
2025-03-14 10:24:51.463 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108379728] SENT: b4 be 00 00 00 0f 01 10 11 88 00 04 08 ff 5e ff ff ff 5e ff ff
2025-03-14 10:24:51.725 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108379728] RECD: b4 be 00 00 00 14 01 10 11 88 00 04 44 dc 00 00 00 00 00 00 00 00 00 00 00 00
2025-03-14 10:24:51.725 DEBUG (MainThread) [pymodbus.logging] Processing: 0xb4 0xbe 0x0 0x0 0x0 0x14 0x1 0x10 0x11 0x88 0x0 0x4 0x44 0xdc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-03-14 10:24:51.725 DEBUG (MainThread) [custom_components.solarman.device] [2108379728] Call 16 ❘ 0x10 ~ 4488 ❘ 0x1188: {'registers': [65374, 65535, 65374, 65535]} failed, attempts left: 4 [error: unpack requires a buffer of 4 bytes]

Invi017, 2025-03-17:
I have LSW-3 and the Storage Control Mode work properly, but i cant change any other option. Discharge or charge power.

[image]

closed 2025-05-01 by github-actions[bot]