davidrapan/ha-solarman #449: Smartload Binary sensor dont working and voltage dont correct

oguzatagan, 2025-03-08:
### Description

hello david. 

Smartload Binary sensor dont working all time seeing off position and voltage dont correct and frequency is 0. Please fix. i m using auto mode for yaml. i attached screenshot

[image]

### Reproduction Steps

i havent

### Log

if you need i will attached

### Version

25.02.14

### Home Assistant Core Version

2025.3.1

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-17:
Hi @oguzatagan,

[fix: Generator Voltage scaling for Deye Hybrid](https://github.com/davidrapan/ha-solarman/commit/711dff0425600d551484bc5e0cabf4a0dbd31764)

And can you please show me value of register number 195 w/ both SmartLoad On and Off using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3)?

Thank you.

oguzatagan, 2025-03-17:
> Hi [@oguzatagan](https://github.com/oguzatagan),
>
> [fix: Generator Voltage scaling for Deye Hybrid](https://github.com/davidrapan/ha-solarman/commit/711dff0425600d551484bc5e0cabf4a0dbd31764)
>
> And can [...]

hi again @davidrapan 

i attached your request screenshot

thanks for all♥️♥️

[image]

oguzatagan, 2025-03-17:
16 is off,  17 is on

davidrapan, 2025-03-17:
[fix: Generator binary sensor in Deye Hybrid](https://github.com/davidrapan/ha-solarman/commit/340ae234c2806a1ee2c90e3090c7ae7be58da8b9)

closed 2025-03-17 by davidrapan