davidrapan/ha-solarman #452: Maximum recursion depth exceeded and HA crashes

Floppe, 2025-03-10:
### Description

Home assistant crashes. Happened twice for me now. Last time in the morning. First device state was "waiting", then "detection" and boom.

[image]

### Reproduction Steps

None. Wait a long dark period when inverter is unreachble and boom when it comes online. I think.

### Log

Exception ignored in: <function StreamReaderProtocol.__del__ at 0x7f1add23aca0>
Traceback (most recent call last):
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line -1, in _reconnect
RecursionError: maximum recursion depth exceeded

### Version

25.02.14

### Home Assistant Core Version

2025.2.5

### Home Assistant Operating System Version

Docker

Floppe, 2025-03-10:
Note that this Sofar Solar inverter goes offline and has no power when there is no sun.

davidrapan, 2025-03-10:
It is most likely solved in the latest commit.