davidrapan/ha-solarman #456: "Setup failed" on v25.03.10

SmolikOne, 2025-03-11:
### Description

Hello, I've just updated Solarman to v25.03.10 and my Integration isn't working anymore. After HA restart I'm receiving error message "Setup failed for custom integration Solarman". Error: setup.py Until previous version everything worked fine. Is there any solution to this issue? 

### Reproduction Steps

Update to new version, no other changes were made. 

### Log

Szczegóły loga (BŁĘDY)
Rejestrator: homeassistant.setup
Źródło: setup.py:340
Pierwsze zdarzenie: 15:28:04 (1 zdarzenia)
Ostatnio zalogowany: 15:28:04

Setup failed for custom integration 'solarman': Unable to import component: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solarman/__init__.py", line 17, in <module>
    from .coordinator import Device, Coordinator
  File "/config/custom_components/solarman/coordinator.py", line 11, in <module>
    from .device import Device
  File "/config/custom_components/solarman/device.py", line 11, in <module>
    from .pysolarman.pysolarman import FUNCTION_CODE, Solarman
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 11, in <module>
    from pymodbus.pdu.decoders import bit_msg, reg_msg, DecodePDU
ImportError: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solarman/__init__.py", line 17, in <module>
    from .coordinator import Device, Coordinator
  File "/config/custom_components/solarman/coordinator.py", line 11, in <module>
    from .device import Device
  File "/config/custom_components/solarman/device.py", line 11, in <module>
    from .pysolarman.pysolarman import FUNCTION_CODE, Solarman
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 11, in <module>
    from pymodbus.pdu.decoders import bit_msg, reg_msg, DecodePDU
ImportError: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)

### Version

25.03.10

### Home Assistant Core Version

2025.3.1

### Home Assistant Operating System Version

14.2

SmolikOne, 2025-03-11:
You are correct, downgrade-upgrade of HA Core and reinstallation of the integration helped and everything is working fine. Thank you!

davidrapan, 2025-03-11:
I'm glad it helped! 😉

closed 2025-03-12 by davidrapan

AdamWeglarz, 2025-03-12:
I still have the problem. I have dowangraded to 2025.02.03 and later to January release. Do I need to completly remove integration and install it again or just for example downgrade/upgrade form HACS?

SmolikOne, 2025-03-12:
Well, the same for me. Everytime I restart HA I need to reinstall Solarman Integration, as it shows errors in the log.

Rejestrator: homeassistant.util.loop
Źródło: util/loop.py:77
Pierwsze zdarzenie: 11:51:06 (1 zdarzenia)
Ostatnio zalogowany: 11:51:06

Detected blocking call to import_module with args ('custom_components.solarman',) in /usr/src/homeassistant/homeassistant/loader.py, line 1074: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 712, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 683, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2040, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 171, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

Rejestrator: homeassistant.setup
Źródło: setup.py:340
Pierwsze zdarzenie: 11:51:06 (1 zdarzenia)
Ostatnio zalogowany: 11:51:06

Setup failed for custom integration 'solarman': Unable to import component: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solarman/__init__.py", line 17, in <module>
    from .coordinator import Device, Coordinator
  File "/config/custom_components/solarman/coordinator.py", line 11, in <module>
    from .device import Device
  File "/config/custom_components/solarman/device.py", line 11, in <module>
    from .pysolarman.pysolarman import FUNCTION_CODE, Solarman
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 11, in <module>
    from pymodbus.pdu.decoders import bit_msg, reg_msg, DecodePDU
ImportError: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
[... cut ...]
  File "/config/custom_components/solarman/coordinator.py", line 11, in <module>
    from .device import Device
  File "/config/custom_components/solarman/device.py", line 11, in <module>
    from .pysolarman.pysolarman import FUNCTION_CODE, Solarman
  File "/config/custom_components/solarman/pysolarman/pysolarman.py", line 11, in <module>
    from pymodbus.pdu.decoders import bit_msg, reg_msg, DecodePDU
ImportError: cannot import name 'bit_msg' from 'pymodbus.pdu.decoders' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py)

AdamWeglarz, 2025-03-12:
and in the same time other modbus integrations for example solaredge modbus are working fine

davidrapan, 2025-03-12:
You will need to report this issue to the Home Assistant repository. 😉

SmolikOne, 2025-03-12:
Ok, submitted. Thx.

SmolikOne, 2025-03-12:
Well, the answer was quick and is pointing back to Solarman issue tracker, meaning this is a issue of this integration... :)

davidrapan, 2025-03-12:
So I'm pointing back to the Home Assistant. :)

SmolikOne, 2025-03-12:
I think they will not hear me :)

SmolikOne, 2025-03-12:
Understood, can I copy your statement to thier issue tracker? Otherwise I will end up just like few minute ago.

davidrapan, 2025-03-12:
Sure.

davidrapan, 2025-03-12:
So according to https://github.com/home-assistant/core/issues/140446 you are guys using some other custom component which also uses pymodbus and forcing HA to download older version of this library which is not compatible w/ Solarman.

**You need to find which one it is. And change it's `manifest.json` or remove it.**

SmolikOne, 2025-03-12:
The only one that I think of is Solaredge Modbus and I think that AdamWeglarz is also using it. So in it's manifest.json we should look for pymodbus and change it to pymodbus>=3.8.3?

davidrapan, 2025-03-12:
Solaredge's Modbus [manifest.json](https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/6a3179cb86063ba92a4457a31ce76ea128d251e7/custom_components/solaredge_modbus/manifest.json#L5) currently contains `pymodbus==3.7.4` so you can first try to change it to `pymodbus>=3.7.4` .

SmolikOne, 2025-03-12:
I've modified the file, and after two host restarts everything still works great. Thank you very much :)

davidrapan, 2025-03-12:
Awesome! 👍

AdamWeglarz, 2025-03-13:
Thanks David. problem is clearly in solaredge modbus integration!