davidrapan/ha-solarman #457: Missing serial configuration

jedifr, 2025-03-11:
### Is your feature request related to a problem?

Hello,

I have a Solis config and my key as a serial number like 5A12338417B0AFF6 (number + letter).
I can't introduce my serial in app configuration...
The app change automatiquely as an other number.

[image]

### Describe the solution you'd like

/

### Describe alternatives you've considered

/

davidrapan, 2025-03-11:
Alphanumeric are not and will not be supported.

davidrapan, 2025-03-11:
Loggers w/ alphanumeric serial numbers are older and completely different hardware. 😉

closed 2025-03-11 by davidrapan

jedifr, 2025-03-12:
Hello, what key do you recommend for a Solis inverter?
Thanks !

davidrapan, 2025-03-12:
[Stick Logger | Solar Data Logger - SOLARMAN](https://www.solarmanpv.com/products/data-logger/stick-logger/)