davidrapan/ha-solarman #460: Integration can't initializing: [FrameError: Fragmented modbus response received]

jkolo, 2025-03-12:
### Description

After update Solarman to version 25.02.10, integration can't initializing. I have Deye Hybrid SG04LP3 Inverter and Ethernet logger LSE3.

### Reproduction Steps

Update Solarman Component to version >= 25.03.10

### Log

2025-03-12 01:33:13.448 WARNING (MainThread) [homeassistant.helpers.service] Referenced entities select.sun_12k_sg04_lp3_work_mode are missing or not currently available
2025-03-12 01:33:25.095 WARNING (MainThread) [homeassistant.helpers.service] Referenced entities switch.sun_12k_sg04_lp3_export_surplus are missing or not currently available
2025-03-12 02:01:50.370 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-10-31T23:58:26.896638+00:00', 'data': {'battery_life_cycle_rating': 6000, 'battery_nominal_voltage': 48, 'inverter_host': '10.38.107.18', 'inverter_mb_slave_id': 1, 'inverter_port': 8899, 'inverter_serial': 2110645731, 'lookup_file': 'deye_sg04lp3.yaml', 'name': 'SUN-12K-SG04-LP3'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JBJG892GNQX5T5R0RYX1VZ2Q', 'minor_version': 6, 'modified_at': '2025-03-12T01:01:40.243300+00:00', 'options': {'host': '10.38.107.18', 'additional_options': {'battery_life_cycle_rating': 8000, 'battery_nominal_voltage': 51, 'mppt': 2, 'pack': 3}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'SUN-12K-SG04-LP3', 'unique_id': 'solarman_2110645731', 'version': 1})
2025-03-12 02:01:50.371 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-03-12 02:01:50.371 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 10.38.107.18
2025-03-12 02:01:50.372 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 10.38.107.18
2025-03-12 02:01:50.375 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [10.38.107.18, 402A8F2CD64B, 2110645731]
2025-03-12 02:01:50.379 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is enabled
2025-03-12 02:01:50.379 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Scheduling 1 query request. ^-1
2025-03-12 02:01:50.379 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 ...
2025-03-12 02:01:50.380 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-03-12 02:01:50.385 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] Successful connection!
2025-03-12 02:01:50.385 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: a5 17 00 10 45 a9 00 e3 e5 cd 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 0d 15
2025-03-12 02:01:51.435 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] TCP_DETECTED: a5 17 00 10 45 03 00 e3 02
2025-03-12 02:01:51.436 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: a5 17 00 10 45 03 00 e3 02
2025-03-12 02:01:51.436 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 4 [FrameError: Incorrect control code]
2025-03-12 02:01:51.437 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: 80 6a 00 00 00 06 01 03 00 00 00 17
2025-03-12 02:01:51.739 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: 80 6a 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2025-03-12 02:01:51.739 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 3 [FrameError: Fragmented modbus response received]
2025-03-12 02:01:51.740 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: e6 e9 00 00 00 06 01 03 00 00 00 17
2025-03-12 02:01:52.042 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: e6 e9 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2025-03-12 02:01:52.043 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 2 [FrameError: Fragmented modbus response received]
2025-03-12 02:01:52.043 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: bd e0 00 00 00 06 01 03 00 00 00 17
2025-03-12 02:01:52.346 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: bd e0 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2025-03-12 02:01:52.347 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 1 [FrameError: Fragmented modbus response received]
2025-03-12 02:01:52.347 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] SENT: 4b ad 00 00 00 06 01 03 00 00 00 17
2025-03-12 02:01:52.649 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2110645731] RECD: 4b ad 00 00 00 31 01 03 2e 00 05 00 01 01 04 32 34 30 35 32 33 36 37 31 33 00 00 00 00 00 00 18 07 d2 07 00 00 20 06 11 44 00 00 10 01 c0 47 00 00 d4 c0 00 01 ff ff
2025-03-12 02:01:52.650 DEBUG (MainThread) [custom_components.solarman.device] [2110645731] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 0, aborting. [FrameError: Fragmented modbus response received]
2025-03-12 02:01:52.650 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18
2025-03-12 02:01:52.652 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-03-12 02:01:52.652 INFO (MainThread) [custom_components.solarman.device] [2110645731] Closing connection to 10.38.107.18

### Version

25.03.10

### Home Assistant Core Version

2025.3.2

### Home Assistant Operating System Version

14.2

voed, 2025-03-12:
Same for me for SG05LP1(deye_hybrid) with modbus connection

hoschilo, 2025-03-12:
Same here HYD 20KTL - 3PH, LSE-3 (Modbus over TCP)

davidrapan, 2025-03-12:
[fix: Detection of modbus response fragmentation for ethernet loggers](https://github.com/davidrapan/ha-solarman/commit/dfa85c656bf0fe19aefc8455bafc8cf76e9c29e6)

davidrapan, 2025-03-12:
[v25.03.12_1](https://github.com/davidrapan/ha-solarman/releases/tag/v25.03.12_1)

hoschilo, 2025-03-12:
solved for me, thanks a lot

closed 2025-03-12 by davidrapan

jandeuss, 2025-03-12:
works for me too. thanks a lot

jkolo, 2025-03-12:
Works for me too. Thanks!