davidrapan/ha-solarman #462: SoC entity problem

Retr0Joe, 2025-03-12:
### Description

Hello, i own 3 Deye RW-M6.1 batteries which are connected to a Deye SUN-12K-SG04LP3-EU inverter. The first two batteries have a problem with the inverter batterie SoC entity since update to 25.03.12_1. Before this version, the batteries did not show the the serial numbers, but SoC was working. Now, the first two batteries show additional information like SN, firmware version etc. but the SoC is not shown anymore, see the home assistant log. The third batterie, where the SN is shown as "0" instead of full SN, shows the correct SoC.

### Reproduction Steps

See description

### Log

State is not JSON serializable: <state sensor.deye_inverter_battery_1=94.2; state_class=measurement, Serial Number=83172650497031628905743619207548036729, MOS=49, Software Version=0000, Hardware Version=0000, unit_of_measurement=%, device_class=battery, icon=mdi:battery, friendly_name=Deye_Inverter Battery 1 @ 2025-03-12T18:26:19.903405+01:00>: Integer exceeds 64-bit range

State is not JSON serializable: <state sensor.deye_inverter_battery_2=94.1; state_class=measurement, Serial Number=49321706587941320675198064230597186420, MOS=49, Software Version=3008, Hardware Version=101056, unit_of_measurement=%, device_class=battery, icon=mdi:battery, friendly_name=Deye_Inverter Battery 2 @ 2025-03-12T18:27:37.988032+01:00>: Integer exceeds 64-bit range

### Version

25.03.12_1

### Home Assistant Core Version

2025.3.2

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-12:
Are you sure that you are running it w/ stock profile? Because it shouldn't generate this long integers after [fix: Battery X Serial Number decoding rule in Deye](https://github.com/davidrapan/ha-solarman/commit/fdd17e87aa3e269849b05a7394da4c5f9a32b23b)

Retr0Joe, 2025-03-13:
Hello @davidrapan, thank you for your response. The profile is manualIy set to "deye_p3". I downloaded the whole files from the repo and copied them manually as described (Copy the contents of custom_components/solarman to /config/custom_components/solarman in Home Assistant) to home assistant, then deleted the devices and readded them. Now all 3 batteries are back with all states and shorter serial numbers. Thank you for your great work!

closed 2025-03-13 by Retr0Joe