davidrapan/ha-solarman #465: Fail to initialize

gpsalt, 2025-03-13:
### Description

After the latest update to 25.03.12, the logger fails to initialize with error: Fragmented modbus response received

### Reproduction Steps

Inverter template: sofar_g3hyd.yaml

### Log

`[36m2025-03-13 09:42:52.369 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-05T11:27:24.727001+00:00', 'data': {'name': 'SofarStick', 'serial': 2108113773}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JEB9681PRFXWZ99KGFKJ57AJ', 'minor_version': 6, 'modified_at': '2025-03-13T08:29:13.591782+00:00', 'options': {'host': '192.168.11.194', 'lookup_file': 'sofar_g3hyd.yaml', 'additional_options': {'mppt': 2, 'pack': 2}}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'SofarStick', 'unique_id': 'solarman_2108113773', 'version': 1})[0m
[36m2025-03-13 09:42:52.369 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh[0m
[36m2025-03-13 09:42:52.369 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.11.194[0m
[36m2025-03-13 09:42:52.370 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.11.194[0m
[36m2025-03-13 09:42:52.372 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.11.194, E8FDF8DB8C65, 2108113773][0m
[36m2025-03-13 09:42:52.372 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: sofar_g3hyd.yaml[0m
[36m2025-03-13 09:42:52.372 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed[0m
[36m2025-03-13 09:42:52.423 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, max_size: 125, digits: 6, attributes: {'mod': 0, 'mppt': 2, 'l': 3, 'pack': 2}[0m
[36m2025-03-13 09:42:52.424 DEBUG (MainThread) [custom_components.solarman.device] {2108113773: {'connections': {('mac', 'e8:fd:f8:db:8c:65')}, 'identifiers': {('solarman', '2108113773')}, 'configuration_url': 'http://192.168.11.194/config_hide.html', 'serial_number': '2108113773', 'manufacturer': 'Sofar', 'model': 'G3HYD', 'name': 'SofarStick'}}[0m
[36m2025-03-13 09:42:52.424 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Scheduling 11 query requests. ^0[0m
[36m2025-03-13 09:42:52.424 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 ...[0m
[36m2025-03-13 09:42:52.426 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] Successful connection![0m
[36m2025-03-13 09:42:52.426 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] SENT: a5 17 00 10 45 d3 00 6d 43 a7 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 73 15[0m
[36m2025-03-13 09:42:52.982 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] TCP_DETECTED: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:52.982 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:52.982 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 failed, attempts left: 4 [FrameError: Incorrect control code][0m
[36m2025-03-13 09:42:52.982 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] SENT: 05 89 00 00 00 06 01 03 04 04 00 28[0m
[36m2025-03-13 09:42:53.325 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] RECD: 05 89 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:53.326 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 failed, attempts left: 3 [FrameError: Fragmented modbus response received][0m
[36m2025-03-13 09:42:53.326 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] SENT: 7c 50 00 00 00 06 01 03 04 04 00 28[0m
[36m2025-03-13 09:42:53.669 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] RECD: 7c 50 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:53.669 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 failed, attempts left: 2 [FrameError: Fragmented modbus response received][0m
[36m2025-03-13 09:42:53.669 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] SENT: 93 f7 00 00 00 06 01 03 04 04 00 28[0m
[36m2025-03-13 09:42:54.002 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] RECD: 93 f7 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:54.002 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 failed, attempts left: 1 [FrameError: Fragmented modbus response received][0m
[36m2025-03-13 09:42:54.003 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] SENT: cb 2f 00 00 00 06 01 03 04 04 00 28[0m
[36m2025-03-13 09:42:54.336 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2108113773] RECD: cb 2f 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 2c 00 00 00 00 00 00 00 00 00 00 00 cf 00 02 1f 4c 00 04 24 15 01 ad[0m
[36m2025-03-13 09:42:54.336 DEBUG (MainThread) [custom_components.solarman.device] [2108113773] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 failed, attempts left: 0, aborting. [FrameError: Fragmented modbus response received][0m
[32m2025-03-13 09:42:54.336 INFO (MainThread) [custom_components.solarman.device] [2108113773] Closing connection to 192.168.11.194[0m
[36m2025-03-13 09:42:54.336 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SofarStick data in 1.912 seconds (success: False)[0m
[36m2025-03-13 09:42:54.336 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown[0m
[32m2025-03-13 09:42:54.336 INFO (MainThread) [custom_components.solarman.device] [2108113773] Closing connection to 192.168.11.194[0m
[33m2025-03-13 09:43:39.211 WARNING (MainThread) [homeassistant.components.sensor] Updating rest sensor took longer than the scheduled update interval 0:00:10[0m
[33m2025-03-13 09:43:39.211 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.mystrom_1 is taking over 10 seconds[0m
[31m2025-03-13 09:43:39.211 ERROR (MainThread) [homeassistant.components.rest.data] Timeout while fetching data: http://192.168.11.51/report[0m
[36m2025-03-13 09:43:54.139 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-12-05T11:27:24.727001+00:00', 'data': {'name': 'SofarStick', 'serial': 2108113773}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JEB9681PRFXWZ99KGFKJ57AJ', 'minor_version': 6, 'modified_at': '2025-03-13T08:29:13.591782+00:00', 'options [... cut ...]
[36m2025-03-13 09:44:40.617 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching SofarStick data in 4.652 seconds (success: False)[0m
[36m2025-03-13 09:44:40.617 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown[0m
[32m2025-03-13 09:44:40.617 INFO (MainThread) [custom_components.solarman.device] [2108113773] Closing connection to 192.168.11.194[0m`

### Version

25.03.12

### Home Assistant Core Version

2025.3.2

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-13:
Duplicate of #460, update to v25.03.12_1

davidrapan, 2025-03-13:
Duplicate of #460

closed 2025-03-13 by davidrapan

gpsalt, 2025-03-13:
Awesome, the fix worked, thanks