davidrapan/ha-solarman #466: LAN Logger with Deye 3PH LV Hybrid inverter not working

trilobyte303, 2025-03-13:
### Description

the solarman integration no longer works after an update to 25.03.12_1

I use a LAN logger and a DEYE SUN-12K-SG04LP3

the logfile is attached. i have tried to re-download the integration, restart etc.

either the error message “Device setup timed out” or “Device setup failed. [FrameError: Incorrect response w/ function code 131 instead of 3 received]”

I have attached a log file.
What can I do to get the whole thing running again?

### Reproduction Steps

install 
Core 2025.3.2
Supervisor 2025.03.2
Operating System 14.2
Frontend 20250306.0

and Solarman intergration 25.03.12_1 on a Raspberry Pi4

### Log

[home-assistant_solarman_2025-03-13T10-42-10.967Z.log](https://github.com/user-attachments/files/19227316/home-assistant_solarman_2025-03-13T10-42-10.967Z.log)

### Version

25.03.12_1

### Home Assistant Core Version

2025.3.2

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-13:
Hi @trilobyte303, you have incorrectly set **Modbus Slave ID** to **3**.

closed 2025-03-13 by davidrapan

trilobyte303, 2025-03-13:
@davidrapan : that is correct, because Modbus ID 3 is set in the inverter:

[image]

davidrapan, 2025-03-13:
But inverter is not replying w/ any data so..

trilobyte303, 2025-03-13:
i configured the solarman integration to modbus adress 1, this works!
(with the modbus id set to 3 in the inverter, after setting to id 1 in the inverter it is still working)

davidrapan, 2025-03-13:
Yes, because it's hardwired/hardcoded to **1** (to my knowledge). I'm glad you were able to sort it out! 😉