davidrapan/ha-solarman #469: Setting Inverter Clock Time incorrectly (5 mins behind correct time)

RaoulSargent, 2025-03-13:
### Description

_First, really pleased to have found this project and the continued development of 'Solarman' integration for Home Assistant.  Thank you and keep up the excellent work._

It appears to be setting the Inverter time on a regular basis (approx every 5 mins) however the time value that it is setting the clock to, appears to be 5 mins behind the current/actual time.

[image]

ha-solarman: 25.03.12
SOLIS INVERTER MODEL: RHI-5K-48ES-5G Hybrid
With six individual Pylontech batteries attached
Home Assistant: Core v:2025.3.2
Platform: DOCKER CONTAINER

This is my test/dev Home Assistant installation so I am happy to install test/debug versions of ha-solarman if I can be of assistance in diagnosing this problem and additional support for Solis inverters.  (I have 2x Solis inverters).

I have checked/confirmed that the date and time are correct on the SYNOLOGY NAS Host, also inside of the Home Assistant Docker Container, and confirmed that the Inverter date/time is incorrectly set when looking at the LCD screen.

Log from Home Assistant attached.

### Reproduction Steps

This is a Docker Container running Home Assistant and ha-solarman (versions as per description text) - I do not know of anything more to 'reproduce' the issue.

### Log

```
2025-03-13 11:32:18.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 11:33:51.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:03:09.765 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:13:51.765 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:20:13.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:28:15.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:30:37.764 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:32:24.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:42:37.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:53:41.764 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 12:59:42.765 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 13:01:21.764 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 13:30:42.764 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data
2025-03-13 13:47:57.843 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-03-13 13:47:58.456 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-03-13 13:47:59.054 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-03-13 13:47:59.652 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-03-13 13:48:01.049 ERROR (MainThread) [pymodbus.logging] Exception response 144 / 0
2025-03-13 13:48:01.050 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139750916931248] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 40, in _async_set_value
[... 178 lines cut ...]
    await self.write(self._to_native_value(value), value.strftime(DATETIME_FORMAT))
  File "/config/custom_components/solarman/entity.py", line 160, in write
    if await self.coordinator.device.exe(self.code, address = self.register, registers = value) > 0 and state is not None:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/device.py", line 129, in exe
    async with asyncio.timeout(TIMINGS_UPDATE_TIMEOUT):
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2025-03-13 13:49:55.763 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Solis1 data

```

### Version

25.03.12

### Home Assistant Core Version

2025.3.2

### Home Assistant Operating System Version

Docker Container

davidrapan, 2025-03-14:
Hi @RaoulSargent, can you please try to do the action w/ debug logging enabled? Thank you.

RaoulSargent, 2025-03-14:
Couple of screen shots from device history log:

[image]

[image]

Debug Log attached.

[home-assistant_solarman_2025-03-14T14-25-42.660Z.log](https://github.com/user-attachments/files/19249171/home-assistant_solarman_2025-03-14T14-25-42.660Z.log)

RaoulSargent, 2025-04-21:
For the benefit of anyone else that may find this issue...  

I have found that the Solis inverter internal clock drifts quite readily meaning any time specific schedule on the inverter for tariffs such as Octopus IGO, Flux, Agile etc, may not align with 'real time'.

_I would assume that this would not be the case where the inverter is able to access the Internet, but in my case my inverter is prevented from doing so and therefore unable to sync with any external time source._

Until ha-solarman is able to sync the inverter clock I have a HA automation that runs at a specific time every day and sets the inverter clock by directly setting the appropriate registers, to the time that the automation runs:

[image]

This is sufficient to keep the inverter clock within 1-2 seconds of real time, and therefore ensure inverter operations tally with the tariff time slots.

RaoulSargent, 2025-05-05:
Hi @davidrapan , this is what my automation is actually doing in terms of setting the registers, which works as intended.

[image]

RaoulSargent, 2025-05-05:
Hi @davidrapan, I am currently using the default supplied Solis Hybrid profile unchanged, which seems to work fine for all other uses... or am I mistaken?

davidrapan, 2025-05-05:
You could try add something like this into the profile:
```
      - name: "Date & Time"
        platform: datetime
        update_interval: 60
        rule: 8
        registers: [43000, 43001, 43002, 43003, 43004, 43005, 43006]
```

davidrapan, 2025-05-07:
[fix: Date & Time registers in Solis](https://github.com/davidrapan/ha-solarman/commit/c3e1dbbca994d2da93446ca81893c09ac093045a)

closed 2025-05-07 by davidrapan