davidrapan/ha-solarman #476: SUN-12K-SG02LP1-EU-AM3 load\CT port bug

fastvd, 2025-03-15:
### Description

Hello. I have a 12kW deye. I connected it to HA, everything seems to be fine with the battery/sun sensors, but there is a problem with the LOAD port - the data is not coming at all...similarly when I connect the load to the inverter via CT connectors...
in the deye cloud the data is correct, but in HA it is not at all coming!
This screen shows how the load from CT is added to the LOAD port. Below is a graph from an external sensor that correctly reads everything that comes after the inverter.
[image]

### Reproduction Steps

I once had an old inverter (similar to PowMr), which does not have a sensor at the input that measures power...there is one at the output, but the data from it was very slow, so I put separate sensors at both the input and output of the inverter, which read the data accurately and quickly. After replacing it with Deye, I left them, so I have a 100% understanding of which data are correct...and unfortunately they do not match!
[image]

Similarly, a very strange situation occurs when we have a sensor that measures the TOTAL power that passes through the inverter, but when my batteries (I have 2 of them, homemade with a JK BMS control board) go on charge, our total power becomes negative!
[image]

### Log

I tried changing the profile type from auto to deye_hybrid, but no change.
At the same time, reading the battery, the sun, and controlling the inverter work well.

### Version

25.03.12

### Home Assistant Core Version

2025.03.2

### Home Assistant Operating System Version

14.2

davidrapan, 2025-03-15:
Hi @fastvd, I'm really not sure if understand... `Power` sensor is the output of the inverter so it can be negative if you are charging batteries from the grid. 😉

fastvd, 2025-03-15:
ok, let it be negative\positive...but when I charge my electric car TO THE INVERTER with the zero export to CT function enabled, this load should not be displayed on the LOAD port!!!
here is the situation when there is no load TO THE INVERTER (CT reads 0)
[image]
and this is roughly the situation we see in the cloud!
[image]

at the moment when nothing is connected to CT, the data from the POWER and LOAD sensors are the same, which is correct and logical...

and now I specifically turned on my electric car for charging and as we can see the data on the LOAD port became the same, although the load should return to CT and be read separately there, which is what is done in the DEYE cloud
[image]

[image]

[image]

P.S.: I forgot to clarify that the data from the LOAD port is not completely correct... if my load is 744 watts, then in HA it will be displayed as 74.4 watts... so I created a separate sensor that multiplies these data by 10 (these are the ones we see on the screenshot in sunsynk-power-flow-card )

[image]

davidrapan, 2025-03-15:
> but when I charge my electric car TO THE INVERTER

What do you mean by that? How it's wired?

fastvd, 2025-03-15:
[image]

fastvd, 2025-03-15:
[image]
I don't fully understand where I can change the parameters of the entities I need here!?

davidrapan, 2025-03-15:
> I don't fully understand where I can change the parameters of the entities I need here!?

Scale modifier will multiply (*10) required sensors .

> I charge my electric car TO THE INVERTER

Okay, and you are talking about discharging your car into the load (**TO** THE INVERTER)? Or is it a typo?

fastvd, 2025-03-15:
> > I don't fully understand where I can change the parameters of the entities I need here!?
> 
> Scale modifier will multiply (*10) required sensors.
> 
> > I charge my electric car TO THE INVERTER
> 
> Okay, and you are talking about discharging your car into the load (**TO** THE INVERTER)? Or is it a typo?

1)no, it's not a typo.
to the inverter = before inverter! google transtate)

2) Scale modifier will multiply (*10) required sensors. - where can I choose the sensor I need?

fastvd, 2025-03-15:
> 1)no, it's not a typo. to the inverter = before inverter! google transtate)
>
> 2. Scale modifier will multiply (*10) required sensors. - where can I choose the sensor I need?

Scale modifier will multiply (*10) required sensors. - I understand...this checkbox really increased the LOAD sensor by 10 times...I just thought that there must be a sensor selection somewhere that shows incorrectly...

fastvd, 2025-03-15:
[image]
For the test, I turned on the battery charge... the POWER sensor became negative, although I don't understand why it should be interpreted that way... what should I do with the load before the inverter?

davidrapan, 2025-03-15:
> although I don't understand why it should be interpreted that way

Because negative indicates power flowing into the device and positive out of the device.

fastvd, 2025-03-15:
ok, maybe there is some logic here)
the External Power sensor should show the power that returns to the CT?

[image]

davidrapan, 2025-03-15:
> the External Power sensor should show the power that returns to the CT?

Or coming from. Again depends on the direction. Positive = coming from, negative = returns to.

fastvd, 2025-03-15:
for the test I added another inverter deye to my HA. my friend has a 5 kW inverter and he also has a load connected in front of the inverter. and I see that the data comes from his inverter correctly! here is the screen!

[image]

fastvd, 2025-03-15:
[image]
and this is my inverter, this is the data without load, everything looks correct, because my whole house is turned on via LOAD PORT

davidrapan, 2025-03-15:
> load connected in front of the inverter

You mean on the Grid/External side?

fastvd, 2025-03-15:
no!
before inverter = on grid home load as Deye calls it! this is a standard connection! there is nothing more standard!
LOAD port = everything that is connected through the inverter itself, here in the picture it is a backup load!
I remind you - on a 6 kW inverter this data is displayed correctly in HA! on my 12 kW this data is NOT DISPLAYED correctly in HA, although at the same time in the deye cloud they come\are displayed\are displayed CORRECTLY!
[image]

fastvd, 2025-03-15:
in deye cloud is true!
[image]

fastvd, 2025-03-15:
and as you can see, when the load goes in front of the inverter, the data from the LOAD PORT also starts to show incorrect!

now you understand that I have not 1 problem, but 2!

moreover, in the cloud deye ALWAYS shows everything correctly! the problem is EXCLUSIVELY in HA!

fastvd, 2025-03-15:
Maybe the issue is somewhere in the config file itself, because in HA my inverter was allocated as SG0XLP1, but I physically have SG02LP1!
[image]

davidrapan, 2025-03-15:
All Deye Inverters (to my knowledge) have stored in the Load register whole Load Power consumption (`Load` + `Backup Load`, mine does that too) and integration only reads those registers. Cloud is proly doing some subtracting (`Load` - `Backup Load` to show `Load Power` consumption w/o Backup Load). It could also depends on the inverter firmware.

BTW, you should have called the ports as they are from the beginning. 😉

> was allocated as SG0XLP1

Sure, **X** is a placeholder for any value 1-5.

fastvd, 2025-03-15:
ok, what sensors should I subtract then to use the Load-Backup Load scheme? 

what to do with the External Power sensor? how is it calculated in the deye cloud, from the sum of which sensors?

davidrapan, 2025-03-15:
> ok, what sensors should I subtract then to use the Load-Backup Load scheme?

I don't know where is `Backup Load` (`Load UPS`) stored in Single-Phase Hybrid Inverters so I'll have to look..

> what to do with the External Power sensor? how is it calculated in the deye cloud, from the sum of which sensors?

`External Power` is not calculated, it's value from CT on grid side.

fastvd, 2025-03-15:
> `External Power` is not calculated, it's value from CT on grid side.

ok, so why is it not showing up in HA for me, but it is showing up in deye cloud?

davidrapan, 2025-03-15:
> ok, so why is it not showing up in HA for me, but it is showing up in deye cloud?

But I can see it on your screenshots..?

> I don't know where is Backup Load (Load UPS) stored in Single-Phase Hybrid Inverters so I'll have to look..

Documentation says nothing about this value, so you will have to look for it on your own using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3)

fastvd, 2025-03-15:
no, let's do everything one by one! you still don't understand what I mean!
Situation number one!---> NOTHING is turned on before the inverter...
1)here is a picture from HA sunsynk-power-flow-card
2) a picture from deye cloud
3) a picture from HA from the device directly with all the entities!

in fact, all data +- coincide (they just arrive more slowly in the cloud of data)
[image]

[image]

[image]

fastvd, 2025-03-15:
Situation number TWO!---> An electric car is turned on before the inverter (4.3 kW for the test)...
1) here is an image from HA sunsync-power-flow-card
2) a picture from a daytime cloud
3) a picture from HA from the device directly with all entities!

as you can see the load from the electric car went to both the general POWER and LOAD POWER, but did not go to our EXTERNAL POWER.
in the deye cloud everything is clearly distributed as it is!
P.S. on the sunsync-power-flow-card where you see the electric train and the number 4.3kW at the bottom right, this is read by a SEPARATE physical sensor, that's why I showed a separate device card with all entities so that it is clearly understood where our data is added\subtracted\not read!

[image]

[image]

[image]

davidrapan, 2025-03-15:
> as you can see the load from the electric car went to both the general POWER and LOAD POWER, but did not go to our EXTERNAL POWER.

Maybe because it shouldn't go into EXTERNAL POWER?! I told you `External Power` = `Grid Power` = 20 W. Which is clearly shown in the cloud too...

fastvd, 2025-03-15:
[image]

davidrapan, 2025-03-15:
I already told you.. Because documentation only provides info about registers for `Load Power` (which is `Load + Backup Load Power`: 1.31 kW + 4.25 kW = 5.56 kW) and doesn't provide info about `Backup Load Power` (which you need for calculation of your `Load - Backup Load Power`) you need to:

> Documentation says nothing about this value, so you will have to look for it on your own using Developer tools >> ACTIONS >> Solarman: Read Holding Registers (Modbus Function Code 3)

magin1, 2025-03-18:
The sensor you need is available but just hidden. sensor.inverter_load_ups_power
Go to device entities, then click on the filter and then Status and then select hidden.

davidrapan, 2025-03-18:
Hi @magin1, `deye_hybrid` profile doesn't have this sensor (yet).

fastvd, 2025-03-18:
Is there a register from this sensor to create it manually?

davidrapan, 2025-03-18:
> Is there a register from this sensor to create it manually?

I already explained it to you here: https://github.com/davidrapan/ha-solarman/issues/476#issuecomment-2726764722

Re4DeR, 2025-03-27:
> > as you can see the load from the electric car went to both the general POWER and LOAD POWER, but did not go to our EXTERNAL POWER.
> 
> Maybe because it shouldn't go into EXTERNAL POWER?! I told you `External Power` = `Grid Power` = 20 W. Which is clearly shown in the cloud too...
> 
> I really don't understand why you think it should affect `External Power` in any way. It's just CT on the grid side, what you are consuming or returning to the grid. You are not consuming power (only 20 W) from the grid nor returning any power to the grid!

Hi! So is there really no difference between External Power and Grid Power? Why are they both listed? Just asking if I should prefer one over the other.

fastvd, 2025-04-03:
> > Is there a register from this sensor to create it manually?
> 
> I already explained it to you here: [#476 (comment)](https://github.com/davidrapan/ha-solarman/issues/476#issuecomment-2726764722)

I searched for registers for a long time. There is no (or I did not find) the necessary data for this inverter. Most likely, in the cloud, deye receives this data by subtracting the total power from something else...
P.S.: and I also do not understand why to make 2 separate sensors for external and CT powers...

768g, 2025-04-03:
Hi guys,

I have migrated form Stephan's version to David's version and on comparing the deye_hybrid.yaml files of each I think we are mssing some entries in the add-on. They do exist in David deye_hybrid.yaml but do not appear to exposed to us in the add-on.

`         Stephan				        Register	 David
sensor.deye external ct l1 power	 0x00AA	 External CT1 Power
sensor.deye internal ct l1 power	 0x00A7	 Grid L1 Power
sensor.deye inverter l1 power	         0x00AD	 Output L1 Power
sensor.deye load l1 power	         0x00B0	 Load L1 Power
sensor.deye total grid power	         0x00A9	 Grid Power
sensor.deye total load power	         0x00B2	 Load Power
sensor.deye total power	                  0x00AF	 Power
`

modenet, 2026-02-08:
how to add  register 185 to my configuration?