davidrapan/ha-solarman #486: Battery capacity  incorrect values

wifi75, 2025-03-19:
### Description

the battery_capacity sensor should show the remaining battery capacity, this value will almost never change, if at all... even if the battery is completely flat or fully charged.

### Reproduction Steps

[image]

### Log

.

[home-assistant_solarman_2025-03-19T11-55-26.567Z.log](https://github.com/user-attachments/files/19340021/home-assistant_solarman_2025-03-19T11-55-26.567Z.log)

### Version

25.03.18

### Home Assistant Core Version

2025.3.3

### Home Assistant Operating System Version

15.0

davidrapan, 2025-03-19:
Hi @wifi75, no, this sensor show **capacity** and not **remaining capacity**.

wifi75, 2025-03-19:
> Hi [@wifi75](https://github.com/wifi75), no, this sensor show **capacity** and not **remaining capacity**.

ok, but my total battery capacity is 30.8 Kwh and not 27.6...

closed 2025-03-21 by davidrapan