davidrapan/ha-solarman #493: Add Off-grid switch

skajper4, 2025-03-22:
### Describe the solution you'd like

Hi. Could you add a switch to toggle offgrid option on deye_p3 config? Below is my working code.

`  - group: Work Mode
    update_interval: 300
      - name: "Off Grid"
        platform: "select"
        rule: 1
        registers: [0x00B3]
        icon: "mdi:electric-switch"
        lookup:
          - key: 0x0000
            value: Auto
          - key: 0x0001
            value: Force`

### Describe alternatives you've considered

.

skajper4, 2025-03-22:
There's no switch in HUD (or I can't find).

strezitom, 2025-04-02:
This does not work for me On a SG04LP12 three phase EU modell, using this _p3 config  :/
I was looking for this functionality for a long time. Made it happen with external single phase switch, cutting one phase and forcing Deye to island.

skajper4, 2025-04-14:
I have SUN-12K-SG04LP3-EU model and it's working perfectly. This option is listed on deye modbus registers list, register 179 bit 2-3

strezitom, 2025-04-15:
If you say its bit 2-3, that does not match your config further up.
I have the exact same model

lookup:
          - key: 0x0000
            value: Auto
          - key: 0x0001
            value: Force

davidrapan, 2025-05-04:
Possibly better solution in #536

davidrapan, 2025-05-08:
[feat: Add Off Grid switch for Deye](https://github.com/davidrapan/ha-solarman/commit/90f65caeb61ed66579b62828d7c7aebff652ed13)

closed 2025-05-08 by davidrapan

vladmihociu, 2025-08-11:
Hello.
I am currently testing the offgrid switch on a Deye SUN-8K-SG01LP1 and I think that it is not working. I am using the auto section while configuring the initial solarman integration setup.
I sepect to see that it stops the export, for example, for that the invertor tension will go from the grid dictated one to the 230 one from grid settings, and it does not.

vladmihociu, 2025-08-12:
> > Hello. I am currently testing the offgrid switch on a Deye SUN-8K-SG01LP1 and I think that it is not working. I am using the auto section while configuring the initial solarman integration setup. I sepect to see that it stops the export, for example, for that the invertor tension will go from the grid dictated one to the 230 one from grid settings, and it does not.
> 
> This sulution doe nothing for me either, on an 12K-SG04 3fase, but the OverFreq 2 setting does work great. I set 52Hz normal, connects to grid, 47Hz Off grid. Not sure what the registers are for that on the SG01, but the same trick should work, as a much more generic Off-Grid switch

Can you please be more specific?
my Hz is pretty stable at 50Hz +- 0.05 I think.

You are setting the value to a greater one, so the inverter will go offgrid because of the mismatch between the set 52 and the read one from grid with is 50?
If yes, then i know that he tries to reestablish conection after a specific period of time that is configured in grid settings I think.