davidrapan/ha-solarman #497: Properly handle all variants of Pylontech Force (all Cell temperatures and voltages)

stheid, 2025-03-23:
### Is your feature request related to a problem?

The current generic pylon tech Force H definition is a good starting point, but it does not properly handle the different variants.

I have a force H3 with 2 modules, but only 60 cells get recognized. looking into the inverter definition shows that those 30 cells of 2 modules are continuously numberd. hence i assume that cell 1-0 (module 1, cell 0) is actually already wrongly numbered.

I can create the three definitions for H1, H2, H3 from the provided generic definition and can test it for H3.
However, i do not know how the automatic selection from the autodiscovery works.

AFAIK, the current generic definition is based on the H2, hence the 30 cells per module (however it should only be up to 5 modules.

I would need help to provide a complete pullrequest.

The configurations from the datasheet are as follows:
- H1: up to 7 modules of 15 cells
- H2: up to 5 modules of 30 cells
- H3: up to 7 modules of 32 cells

### Describe the solution you'd like

Autodiscover and correctly handle all Pylontech Force variants

### Describe alternatives you've considered

Solve it manually myself

davidrapan, 2025-05-04:
Hi @stheid, so did you tried to modify the file and add Cells number 31 and 32?

stheid, 2025-05-05:
I didnt do it because personally i am not that interested in all temperature and voltage values, min and max suffices for me.

I can test it out if there is interest in generally improving the handling of the Force H1, H2, and H3 devices.

I would also suggest to disable all individual cell voltage and temperature sensors per default (or provide a flag in the configuration to hide them.) adding over a hundred sensors seems a bit overkill if there is also a min-max voltage and min-max temperature sensor.

stheid, 2025-05-05:
Changing the "inverter definition" to read the 64th temperature value (module 1, cell 31 (0-indexed)) certainly produces a reasonable value.

So i think it would work properly.

davidrapan, 2025-05-12:
[refactor: Add missing modules and sensors to Pylontech Force](https://github.com/davidrapan/ha-solarman/commit/3990b767ac3649296aa0ab710e4073b5e91d3061)

stheid, 2025-05-13:
Hi, i don't think the version is correct now. Now, some registers are read multiple times.
https://github.com/davidrapan/ha-solarman/blob/3990b767ac3649296aa0ab710e4073b5e91d3061/custom_components/solarman/inverter_definitions/pylontech_Force-H.yaml#L670-L676 ↑ 0-31 for the Force H3 (last value module 1)
↓ 1-1 for the Force H2 (second value module 2)
https://github.com/davidrapan/ha-solarman/blob/3990b767ac3649296aa0ab710e4073b5e91d3061/custom_components/solarman/inverter_definitions/pylontech_Force-H.yaml#L947-L953 .. 2-1 for the Force H1 (second value module 3)

As said, the values seem to be stored continuously, which means that the different versions need to be handled differently as they have different amounts of cells per module.

🐛 Actually, the configuration is broken now and doesn't work anymore for me.

davidrapan, 2025-05-13:
> Actually, the configuration is broken now and doesn't work anymore for me

You will have to tell me more than that...

davidrapan, 2025-05-13:
[fix: Remove overlapping sensors in Pylontech Force](https://github.com/davidrapan/ha-solarman/commit/8bb558099fa61e16d64bc6aa68ac16e555ea5b90)

stheid, 2025-05-13:
I think the right way would be to create 3 configs, one for each version of the Pylontech Force Hx. The number of cells are different for every variant and there is not correct way to enumerate them on all devices.

Sorry, i have not found the time to investigate the bug. I just wanted to mention as i assume its because of the overlapping registers.

Do you have a script to generate those large config files? otherwise i might contribute one and generate the correct registor lists for all 3 variants. just let me know what you think.

davidrapan, 2025-05-13:
This is useless when we don't even know the register addresses. H3 may have 2 more cells than H2, but there is no room left between the currently used registers, so who knows where they are and number of modules/packs is configurable.

davidrapan, 2025-05-13:
Ah, I see. I wouldn't have guessed that it would be solved this way. But in this case I really have no idea why the profile didn't work for you then. :) Because multiple register redefinition is allowed.

stheid, 2025-05-14:
So i think there was no bug (just the router decided to give the battery a new IP address again 🙄)

I am running the 25.05.12 version. However the data still doesnt seem to be very plausible to me.
[image]

Does the configuration automatically detect the variant of the Battery (H1,H2,H3)? Do i need to reconfigure or would we have to create different configurations for each variant? How can i help?

As i said previously, i would still love if the configurator had an option to deactivate all individual cell readings during configuration because it creates more than 100 sensors that might not be important for most of the people, but maybe this is a discussion for another issue.

davidrapan, 2025-06-04:
[feat: Add support for all three configurations H1, H2 and H3 of Pylontech Force](https://github.com/davidrapan/ha-solarman/commit/7fae1d4f79b65d8f3dfe657997cd62b9aabe1e85)

closed 2025-06-04 by davidrapan

stheid, 2025-06-06:
I had to edit the "modifier" manually, but probably thats a migration issue. The values are plausible now. Amazing work. thanks.

[image]

davidrapan, 2025-06-06:
What did it do? Loaded profile w/ default modifier value?

stheid, 2025-06-06:
i didnt want to delete my config, so i edited the existing device configuration (where i needed to set the new yaml file) but it did not automatically select the modifier. i had to go to the extended options and correct it. also "modifer" is not a very good name maybe, expecially because its 0-indexed...

[image]

davidrapan, 2025-06-06:
Oh yeah the migration only works w/ the default profile and modifier is universal for many other profiles, that’s why. But I should have made it a selector. You’re right. ;)