davidrapan/ha-solarman #500: Inverter Disconnects

ckoller88, 2025-03-26:
### Description

Hello, 

I have 2 10KW Deye hybrid inverters in master slave configuration, When I set them up the slave works perfectly fine, but the master disconnects and reconnects about every 10s, that leads to the data not being available to the solarman app anymore. The Logger itself remains available both in the app and over the webpage, so i am not sure if this is a software problem or hardware issue with the inverter. I also tried the StephanJoubert Project in Home assistant and the disconnections do not occur there, however the values presented are not usable.

### Reproduction Steps

2 Deye hybrid inverters in master slave mode with Solarman Sticks, connected over wifi. Add them in Home assistant, Slave will stay connected, Master will disconnect and reconnect every 5-10 seconds.

### Log

2025-03-24 23:01:51.310 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data
2025-03-24 23:01:51.361 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_today_energy_import, value 2.3 with device class None and unit of measurement kWh excluded from calculation in sensor.imported_energy_daily
2025-03-24 23:01:51.361 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_today_energy_import with value unavailable excluded from calculation in sensor.imported_energy_daily
2025-03-24 23:01:51.362 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_today_production, value 19.5 with device class None and unit of measurement kWh excluded from calculation in sensor.yield_energy_daily
2025-03-24 23:01:51.363 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_today_production with value unavailable excluded from calculation in sensor.yield_energy_daily
2025-03-24 23:01:51.363 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_battery_power, value 458 with device class None and unit of measurement W excluded from calculation in sensor.zero_power
2025-03-24 23:01:51.364 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_battery_power with value unavailable excluded from calculation in sensor.zero_power
2025-03-24 23:01:51.364 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_load_power with value unavailable excluded from calculation in sensor.inverter_load_power_2
2025-03-24 23:01:51.364 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_load_power, value 468 with device class None and unit of measurement W excluded from calculation in sensor.inverter_load_power_2
2025-03-24 23:01:51.365 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_pv_power, value 0 with device class None and unit of measurement W excluded from calculation in sensor.inverter_power
2025-03-24 23:01:51.366 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_pv_power with value unavailable excluded from calculation in sensor.inverter_power
2025-03-24 23:01:51.373 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_grid_power, value 121 with device class None and unit of measurement W excluded from calculation in sensor.feedin_power
2025-03-24 23:01:51.373 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_grid_power with value unavailable excluded from calculation in sensor.feedin_power
2025-03-24 23:01:51.374 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_grid_power, value 121 with device class None and unit of measurement W excluded from calculation in sensor.meter_power
2025-03-24 23:01:51.374 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_grid_power with value unavailable excluded from calculation in sensor.meter_power
2025-03-24 23:01:51.375 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.feedin_power with value unavailable excluded from calculation in sensor.khb
2025-03-24 23:07:53.308 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data
2025-03-24 23:07:53.339 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_today_energy_import, value 2.3 with device class None and unit of measurement kWh excluded from calculation in sensor.imported_energy_daily
2025-03-24 23:07:53.339 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_today_energy_import with value unavailable excluded from calculation in sensor.imported_energy_daily
2025-03-24 23:07:53.340 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_today_production, value 19.5 with device class None and unit of measurement kWh excluded from calculation in sensor.yield_energy_daily
2025-03-24 23:07:53.340 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_today_production with value unavailable excluded from calculation in sensor.yield_energy_daily
2025-03-24 23:07:53.341 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_battery_power, value 487 with device class None and unit of measurement W excluded from calculation in sensor.zero_power
2025-03-24 23:07:53.341 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_battery_power with value unavailable excluded from calculation in sensor.zero_power
2025-03-24 23:07:53.342 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_load_power with value unavailable excluded from calculation in sensor.inverter_load_power_2
2025-03-24 23:07:53.342 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_load_power, value 487 with device class None and unit of measurement W excluded from calculation in sensor.inverter_load_power_2
2025-03-24 23:07:53.343 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_pv_power, value 0 with device class None and unit of measurement W excluded from calculation in sensor.inverter_power
2025-03-24 23:07:53.343 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only numerical states are supported, entity sensor.inverter_pv_power with value unavailable excluded from calculation in sensor.inverter_power
2025-03-24 23:07:53.351 WARNING (MainThread) [homeassistant.components.group.sensor] Unable to use state. Only entities with correct unit of measurement is supported, entity sensor.inverter_2_grid_power, value 118 with device class None and unit of measurement W excluded from calculation in sensor.feedin_power
[... cut ...]
2025-03-24 23:09:57.683 DEBUG (MainThread) [custom_components.solarman.parser] Value: -100.0 of battery_temperature_sensor validation failed: {'min': -99, 'max': 99}
2025-03-24 23:09:57.684 DEBUG (MainThread) [custom_components.solarman.device] [2708628350] Returning 67 new values. [Previous State: Connected (1)]
2025-03-24 23:09:57.685 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter 2 data in 0.337 seconds (success: True)

### Version

25.02.14 (I also tried the latest version, problem is the same)

### Home Assistant Core Version

2025.3.4

### Home Assistant Operating System Version

15.0

rtripon, 2025-04-20:
Hello, 
I think I'm having the same or similar problem, even 'tho I have only 1 inverter(10kW 3p version).
It keeps disconnecting every 20-30 seconds, didnt have this problem few versions ago.

[solarman disconnecting.txt](https://github.com/user-attachments/files/19824278/solarman.disconnecting.txt).

Version
25.03.18 - Latest

Home Assistant Core Version
2025.4.3

Home Assistant Operating System Version
15.2

davidrapan, 2025-05-04:
Hi @ckoller88 and @rtripon, can you both verify/confirm that you have a stable wifi connection to the inverters? If you can ensure that try to downgrade integration version to v24.12.22 and see if anything changes...

ckoller88, 2025-05-04:
Hi @davidrapan,

Thank you looking into this. 
The wifi connection should be stable, since the other inverter right next to it has a stable connection and also my mqtt dongle i‘m using now for the problematic inverter. 
I tried the current version and also downgraded to 24.12.08. with the same results.

davidrapan, 2025-05-04:
So you have two inverters. One is working just fine. The other not so much? What are firmware versions on both your inverter's logger? Are they the same?

ckoller88, 2025-05-04:
yes I have two identical 10KW Three phase LV Hybrid inverters running, 
the one that's not working is the "Master", the solarman app shows for the logger: Module Version No：LSW3_15_FFFF_1.0.84, Extended System Version：V1.1.00.0F
The one that's working is: Module Version No：LSW3_15_FFFF_1.0.91R, Extended System Version：V1.1.00.0F.

davidrapan, 2025-05-04:
So the culprit is obvious, faulty firmware. Contact solarman and ask them to upgrade LSW3_15_FFFF_1.0.84 to LSW3_15_FFFF_1.0.91R and  you will be just fine. 😉

ckoller88, 2025-05-04:
Oh wow thank you! can't Believe i haven't checked it yet.. I'll see if i can upgrade and report when i have results!

rtripon, 2025-05-05:
> Hi [@ckoller88](https://github.com/ckoller88) and [@rtripon](https://github.com/rtripon), can you both verify/confirm that you have a stable wifi connection to the inverters? If you can ensure that try to downgrade integration version to v24.12.22 and see if anything changes...

Hello,
My wifi is strong table, I have an AP in the same room with the inverter, so no issues there. 
I've just looked at logs on both last version and v24.12.22(I have 2 homeassistant servers, one in my basement where inverter is and one in a remote location connected via VPN).
Both showed disconnects at the same time, and 'tho are not very often, are annoying because its spamming my logs with information I dont want.
These disconnects happens 5-10 times/day at random hours: last night at 11:50, this morning at 3, at 6 and at 8 A.M.
Firmware version : LSW3_15_FFFF_1.0.96 wich is the one dongle came with last year.(didnt have any issues untill few months ago so I didnt feel like making any updates since everything was running ok).
Dont know for sure, but it seems that since I put the dongle on fixed IP, the disconnects happens less.

davidrapan, 2025-05-05:
How long are the disconnects? It doesn't look like the problem is in the integration though. There is simply no response from the logger. Even so it should work just fine with multiple instances running it could potentialy overwhelm the logger when requests from both instances meet in time so try run just one for some time and see if anything changes.

rtripon, 2025-05-07:
Disconnects takes like 1 second, I see them only in logs and  arent affecting my dashboards at all.
I stopped the second HA server and will  monitor the logger to see if the issue persists.
So far only once since yesterday 5 P.M.

davidrapan, 2025-05-09:
Wow, that's.. hm.. 😆 weird.

ckoller88, 2025-05-09:
Firmware has been updated to LSW3_32U_5406_1.0B, seems to have fixed it! 
Thank you for your help!

sofkaski, 2025-05-12:
This might be irrelevant information if there is a stick logger firmware that works with ha-solarman, but for the record....

My logger firmware is LSW3_32U_5406_1.06 and it has issues. I have seen lot of connection problems, but fortunately ha-solarman recovers quite nicely most of them, but some cause momentary unavailability a couple of times a day (with which I have been able to live).

I did today an attempt to understand what really happens on the TCP level in logger communication. I used tcpdump/wireshark + put solarman to debug logging  mode at the same time.  

The [attached text file](https://github.com/user-attachments/files/20171138/Error.cycle.txt)
 contains annotated solarman debug log snippets + wireshark summaries. It contains one sunny day update sequence (success) and two failure sequences where solarman is able to recover and update successfully.

Unfortunately I have not been able to catch (yet) the sequence where an update round fails with `ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Inverter data`. This happens a couple of times a day.

## The normal sunny day sequence:

In this sequence the connection already exists.

  1. Solarman sends a request to the Logger
  2. Logger acknowledges and sends a response
  3. Solarman acknowledges the response and processes it
  
  The sequence takes less than half a second.

## Failure sequence 1

In this failure sequence:
  1. Solarman sends a request
  2. Logger does not respond to a request and Solarman resends the request 5 times
  3. Logger resets the connection
  4. Solarman re-connects successfully and resends the request
  5. Logger responds
  6. Solarman acknowledges the response and processes it successfully

  
The sequence takes about 12 seconds. This failure sequence happens randomly a few times a day.

 ## Failure sequence 2

In this failure sequence:
  1. Solarman sends a request
  2. Logger does not respond to a request and Solarman resends the request 5 times
  3. Solarman closes the connection
  4. Solarman attempts to re-connect, but the Logger refuses (resets the connection) (this repeats ~50 times)
  5. Finally Solarman re-connects successfully and resends the request
  5. Logger responds
  6. Solarman acknowledges the response and processes it successfully
  
The sequence takes about 18 seconds. This failure sequence happens about every 30 minutes.

davidrapan, 2025-05-12:
I'm also running mostly LSW3_32U_5406_1.06 but w/ 0 disconnects. Weeks of persistent connection.. so it's most likely not about the firmware.. but then, I don't know what else it could be..

I'm also closing the thread because the problem was solved by updating the second inverter also to LSW3_32U_5406_1.0B

closed 2025-05-12 by davidrapan

sofkaski, 2025-05-13:
> I'm also running mostly LSW3_32U_5406_1.06 but w/ 0 disconnects. Weeks of persistent connection.. so it's most likely not about the firmware.. but then, I don't know what else it could be..
> 

Configuration of the stick is one possibility.  Main items from my configuration (from `config_hide.html`):

* working mode: data collection
* server A setting: (empty)
* optional server setting: (empty)
* Internal server parameters setting, TCP time out setting: 300
* Mode setting: AP/STA

davidrapan, 2025-05-13:
I mean these aren't really options which could change anything but I have it the same. 😉 The only difference is that I now have it connected to the soalarman servers, so the settings for server A are filled in.

dmagyar, 2025-05-22:
I'd love to update to LSW3_32U_5406_1.0B but I can't find the firmware. Could anyone please upload it to somewhere? I'm plagued by these disconnections :(

davidrapan, 2025-05-22:
Hi @dmagyar, write to solarman supp and ask them to update your stick to this version. 😉

sofkaski, 2025-05-27:
I installed `25.05.26`. I'll keep an eye on unrecoverable communication errors. None so far.  I'll update the results  in  [release note discussion](https://github.com/davidrapan/ha-solarman/discussions/619#discussion-8374428)

gzdr, 2025-07-09:
Hi
I have readed a lot of article and tried many things for solved my issue with the solarman integration but still no lucky:
My issue is similar, but have no idea how to fixit. Installation looks like below
Logger: Sollarman LSW3 with the fresh firmware: LSW3_15_1E03_1.33 / Data transfer cycle：5Min
Inverter: Afore - HNS2000TL-1 single phase V2.15Build300

I tried adding this integration but still no success, is some one who can help my understand what I'm doing wrong?
Should I increase socket timeouts? where exactly?
Reduce Query Load?

I added two log files, one for TCP and second for Modbus-TCP
I will be very grateful for your help

[TCP-home-assistant_solarman_2025-07-09T11-11-52.260Z.log](https://github.com/user-attachments/files/21142293/TCP-home-assistant_solarman_2025-07-09T11-11-52.260Z.log)
[TCP-Modbus-home-assistant_solarman_2025-07-09T11-44-24.602Z.log](https://github.com/user-attachments/files/21142294/TCP-Modbus-home-assistant_solarman_2025-07-09T11-44-24.602Z.log)

gzdr, 2025-07-17:
anyone? some ideas?

sofkaski, 2025-07-17:
> anyone? some ideas?

Logs do not show why the connection is suddenly closed after a couple of successful queries. I have used `tcpdump` to investigate what happens on TCP level between the HA instance and the logger stick. I wrote some time ago [instructions](https://gist.github.com/sofkaski/4c329aff12a74e53c258a56cace1db7a) about it.  That dump and `wireshark` may give more insight why the connection is abruptly closed.

gzdr, 2025-07-20:
Hi, I saw your article. I thought the problem was known and could be quickly resolved. I'm trying to collect Wireshark dumps and investigate what's wrong.
Thanks

Jevgenijs-JG, 2026-02-10:
Good day!
Today in the morning suddenly inverter start to become Unavailable.

[image]

Logger:
Module Version No：LSW3_32U_5406_1.0B - last version.
Data Acquisition Period：60s
Data Uploading Period：5Min
Signal Strength：100

Made Core Rebuild in terminal - ha core rebuild  - not help.
Router restarted.
Please advise, what else I can do.

Jevgenijs-JG, 2026-02-10:
Please advise me, where to check period time, when integration connects to logger. Possible increasing this time period will help