davidrapan/ha-solarman #503: Autodetect fails on Sofar ME3000SP

peterdedecker83, 2025-03-27:
### Description

Hi,

Just to notice: upon installing this integration, the configuration process automatically detected the correct IP address and SN. Profile was set to "Auto" but that didn't work out and I got this error:

pluginFailed setup, will retry: [2104729773] Device setup failed. [FrameError: Incorrect response w/ function code 131 instead of 3 received]

When I opt for the sofar_wifikit.yaml, configuration went smooth and all entities were created and readings are fine.

My inverter model is SE1ES430, firmware version of the stick is ME_0D_2701_5.09. The stick is connected over wired ethernet to my router.

### Reproduction Steps

Install the plugin, configure, verify SN and IP, set profile to Auto.

### Log

[home-assistant_solarman_2025-03-27T12-45-00.983Z.log](https://github.com/user-attachments/files/19486606/home-assistant_solarman_2025-03-27T12-45-00.983Z.log)

### Version

25.03.18

### Home Assistant Core Version

2025.3.4

### Home Assistant Operating System Version

15.0

davidrapan, 2025-04-03:
Hi @peterdedecker83, list of devices w/ profile auto detection is available [here](https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices).

closed 2025-04-03 by davidrapan

peterdedecker83, 2025-04-08:
OK, didn't see that one, thanks!