davidrapan/ha-solarman #505: Missing grid power consumption entity

mtshare, 2025-03-27:
### Description

Hi,
first of all, thanks for the great work on this integration!

I’ve successfully configured the ha-solarman integration in Home Assistant and I’m able to see most of the values correctly. However, I’m not seeing any entity related to power consumption from the grid.
Whether I’m importing or exporting energy, there’s no entity showing this value.

Interestingly, if I use the official Solarman app, I can see the grid power consumption value. But when I use the ZCS app, that specific value is missing — maybe this is related?

For reference, my setup includes:
	•	Logger: Stick Logger LSW-3-C
	•	Inverter: ZCS 3PH HYD10000

Is this something that could be added to the integration, or is it a limitation from the API/data source?

Thanks in advance for your help, and let me know if you need any logs or further info!

### Reproduction Steps

[image]
[image]

### Log

[home-assistant_solarman_2025-03-27T18-11-00.704Z.log](https://github.com/user-attachments/files/19491897/home-assistant_solarman_2025-03-27T18-11-00.704Z.log)

### Version

25.03.18

### Home Assistant Core Version

2025.3.4

### Home Assistant Operating System Version

15.0

ccfiel, 2025-04-30:
how did you solve this issue?

davidrapan, 2025-05-04:
Hi @mtshare, what profile are you using?

closed 2025-06-18 by github-actions[bot]