davidrapan/ha-solarman #509: Not working with Deye Data Logger (Deye Cloud)

macreyes1, 2025-03-29:
### Is your feature request related to a problem?

SUN-6K-SG04LP1-EU-SM2 this is my inverter. Would you happen to have a solution? 

I can seem to configure properly to get detected

### Describe the solution you'd like

I dont have a solutio 

### Describe alternatives you've considered

None

strezitom, 2025-04-01:
The initial issue is quite vague, however I have something similar. Since the integration is running I can no longer control the unit from Deye Cloud as an installer, read command times out ...
The communication to DeyeCloud is fine (every five minutes) and the graphs are uninterrupted.

macreyes1, 2025-04-02:
Sorry I wasnt clear.
I'm unable to configure HA to connect to the inverter. 
I tried the 2 existing integrations (one says it fails to connect) this one shows the entities but without data (not connected to the inverter)

davidrapan, 2025-04-03:
Hi @macreyes1, your device is supported, but if you are having any issues, I will need you to provide me with a debug log.

macreyes1, 2025-04-04:
Does it connect to tcp 8899?
My logger tcp8899 port is not open. 

I'm able to connect to the inverter via deye cloud app though.

lgiganteb, 2025-04-04:
hey David, 

I have the same issue with a HIS 5000/48 (brand is Turbo Energy). When I enable the integration using the latest version (25.03.18), their app stops collecting data. I'm happy to attempt getting youu the debug log, but can you guide me how to do it?

I tried to use older versions as I've been told in a solar installations forum it works, but they fail to initialize the inverter for some reason.

davidrapan, 2025-04-04:
Yes the integration uses tcp port number 8899. Maybe you are running one of the firmware version w/ this port disabled?

davidrapan, 2025-06-06:
You can always ask for a downgrade.

h1771t, 2025-06-07:
Just for info purposes, nc always shows udp as success, rather use nmap.
nmap on my stick shows tcp 80 open, and udp 53. Nothing else is open.
I may attempt going to modbus route to pull the inverter data to HA, using https://esphome.io/components/modbus_controller

h1771t, 2025-06-07:
Thanks I will ask my installer and see if they can downgrade stick firmware for me.
`http://IP_ADDRESS` gives me a login page with "Deye Device Configuration" heading. `admin/admin` gives me incorrect password.
Weirdly `http://IP_ADDRESS/config_hide.html` gives me just firefox connection was reset on multiple tries. It doesn't even give me 404, just straight firefox `NS_ERROR_NET_RESET`
On the physical stick itself, it says `name: Deye data logger` and `model: DL1000B-WIFI` The logger stick as two blue LEDs labelled `NET` and `COM`
The "Deye cloud" android app successfully added the logger after QR scan of the logger.

davidrapan, 2025-06-07:
Hmmm, there is veeeeeeery little info about this logger. It really might be w/o direct Modbus access support all together. Deye is really going backwards w/ the "development" if that's the case, really. 😞

h1771t, 2025-06-07:
I will ask my installer if it's possible to swap out and give me a stick logger with modbus tcp on it at all. Let me see what they say. I suspect they'll refuse given that this is the stick that came with the inverter, but worth a try.

Rik-vdM, 2025-07-12:
Hi David,

I'm afraid that the DL-1000B-WIFI data logger is going to be the future for Deye inverters. On the newest inverters the port through which the data logger is connected has been changed from a DB9 connector to a USB connector. Consequently the older LSW-3-C data logger can no longer be used with these newest inverters. The board inside the inverter has also been changed, so there is probably no way to "down grade" to the old logger.

The DL-1000B-WIFI data logger works fine with the Deye Cloud, it seems to pass all the same data that the LSW-3-C data logger does. One change they have made is that on the first logon to the web server in the data logger you are forced to change the password, and admin/admin is no longer accepted as the password needs to be at least 6 characters.

Let me know if there is something you would like me to test. I have inverters with both types of data loggers installed.

davidrapan, 2025-08-09:
Solarman has loggers with USB too but who knows what's going through with this newer implementation.. you can either look around the newer logger and try to find a way for loacl connections (if there is any) or ask support what are your options for local connections.

Depends on what's inside but there is always a way. 😉

midnextlevel, 2025-09-09:
wow im actually using this DL1000B-WIFI installed by our installer and insisted to use it, anyway to hack it?

closed 2025-10-25 by github-actions[bot]