davidrapan/ha-solarman #511: Deye / tewaycell 16s: discharge battery mode

TonKro, 2025-03-30:
Is there some more documentation available?

Enable  Charging is working fine but how can I set the battery into discharge mode? And how to control the discharge power?

davidrapan, 2025-04-03:
Hi @TonKro, I'm not sure exactly what you mean, but generally you would use Time of Use Programs to manage (dis)charging.

TonKro, 2025-04-04:
Thanks for your message.

Can you help me by hand of an example how to discharge / sell to grid?

greenMikeEU, 2025-04-09:
Hi @TonKro and @davidrapan 
Using the display of my inverter, I can change the necessary values so that the battery can discharge to the grid. That works fine so far.

[image]

"Selling First" must be enabled

"Max Sell Power" defines the amount of power (in watts) that gets exported to the grid

The checkbox for "Grid Peak Shaving" must be disabled

In Home Assistant, I can't find the switch for "Grid Peak Shaving". I looked into the GitHub repo of the Solarman integration and I can find the relevant definition:

Do you (@davidrapan ) have any idea why I can't find the switch?

File:
https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/deye_hybrid.yaml
Line 2167
- name: Grid Peak shaving
  platform: "switch"
  rule: 1
  registers: [0x0118]  # Register 280
  value:
    bit: 8

Here all my switches from Deye in Homeassistant

[image]

Thx for your help

TonKro, 2025-04-09:
It is not there indeed.

Can you maybe help to understand how to charge only from solar? There is a setting to charge from GRID but even even when disable this in HA the inverter keeps charging? My Battery is now 25%, maybe there is a setting on this?

greenMikeEU, 2025-04-09:
I can send you my settings I don't charge anything from the grid only via the PV modules

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TonKro, 2025-04-09:
Great, going to test.

greenMikeEU, 2025-04-13:
Did it work?

davidrapan, 2025-05-08:
Hi @greenMikeEU, version with those changes was released today.

@TonKro, you don't have System Work Mode page in your inverter settings? That's very unlikely.