davidrapan/ha-solarman #513: Errors in logs in HA

000al000, 2025-04-02:
### Description

Errors in logs in HA continiously repeated

### Reproduction Steps

Install solarman integration
Use with deye 12k 3 phase + 8 batteries deye, profile deye_p3.yaml
Version of component -  25.03.18

### Log

Logger: homeassistant
Source: custom_components/solarman/sensor.py:114
integration: Solarman ([documentation](https://github.com/davidrapan/ha-solarman), [issues](https://github.com/davidrapan/ha-solarman/issues))
First occurred: March 31, 2025 at 9:50:30 AM (5467 occurrences)
Last logged: 9:35:14 AM

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/solarman/entity.py", line 77, in _handle_coordinator_update
    self.update()
    ~~~~~~~~~~~^^
  File "/config/custom_components/solarman/entity.py", line 87, in update
    if (data := self.coordinator.data.get(self._attr_key)) is not None and self.set_state(*data) and self.attributes:
                                                                           ~~~~~~~~~~~~~~^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 114, in set_state
    if self._sensor_ensure_increasing and self._attr_native_value and self._attr_native_value > state > 0:
                                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>' not supported between instances of 'float' and 'NoneType'

### Version

25.03.18

### Home Assistant Core Version

2025.3.4

### Home Assistant Operating System Version

15.0

haysdb, 2025-04-02:
Is the integration working? Just errors in the log?

000al000, 2025-04-02:
Yes, working. Errors in log repeats once every 30 seconds

plebann, 2025-04-02:
It's hard to confirm that it's working 100%, recently I'm seeing quite some amount of issues with refreshing sensors (like generator production or AC/DC temperature) so there might be some connection, hard to say really.

JFHCFC, 2025-04-03:
The plugin is fantastic.

I have a problem with the entity
sensor.inverter_pv_power

it shows power 10 times higher than the actual

[image]

davidrapan, 2025-04-03:
Hi @000al000, can you enable debug logging and attach the file? Thanks.

> I have a problem with the entity: `sensor.inverter_pv_power`

Hi @JFHCFC, can you tell me which profile this is?

JFHCFC, 2025-04-03:
Profile deye_sg04lp3.yaml

i've got SUN-10K-SG04LP3-EU

davidrapan, 2025-04-03:
Make sure you have not accidentally enabled scale modifier in additional options of configured device.

plebann, 2025-04-03:
I have the same error, I'll attach my log too.

[home-assistant_solarman_2025-04-03T17-21-31.241Z.log](https://github.com/user-attachments/files/19591915/home-assistant_solarman_2025-04-03T17-21-31.241Z.log)

JFHCFC, 2025-04-04:
> Make sure you have not accidentally enabled scale modifier in additional options of configured device.

thank you,
after unchecking 'scale modifier' it shows correctly

[image]

000al000, 2025-04-04:
> Hi [@000al000](https://github.com/000al000), can you enable debug logging and attach the file? Thanks.

[home-assistant_solarman_2025-04-04T06-14-12.594Z.log](https://github.com/user-attachments/files/19599093/home-assistant_solarman_2025-04-04T06-14-12.594Z.log)

opensource4life, 2025-04-18:
I have exactly the same error as 000al000. Probably relevant: I have a LSE-3 ethernet logger (COM version, not USB) with firmware ME_0E_5406_4.0C.

davidrapan, 2025-05-04:
[fix: Comparison of incomparable types](https://github.com/davidrapan/ha-solarman/commit/350a21b0915936ed9bb6e2f85e324a042f099cac)

closed 2025-05-04 by davidrapan