davidrapan/ha-solarman #516: Partially offline in DeyeCloud when the integration is enabled

Stormhand, 2025-04-04:
Hello,

I have deye 8kw SG01LP1 with Solarman LSW3 and just found this integration. When installed it i found that in the Deye UI the status switches from Online to Partially offline immediately if the integration is enabled and goes back to Online when i disable it. I not sure exactly how it affects the cloud and what "partially" means.
Am i supposed to use both in the same time or have to choose one of them. If its a matter of configuration, can i change some parameter to make them work together?
I had one more issue but not sure if its related, the night after i installed ha-solarman the configuration in "System Work Mode-2 " which should charge the batteries on night tariff "Grid Charge" didnt trigger.

Thanks!

david9600, 2025-04-13:
I have a similar (probably the same) issue with an LSW-3 (on Sofar inverter).
It appears that when the HA integration connects to the logger, the logger's ability to upload data to the cloud is impacted.
I've set up an environment to take some Wireshark traces to help understand what's going on.
I've also emailed SolarmanPV, asking for confirmation of current firmware for the logger, in case my version isn't allowing concurrent TCP connections.
Since each platform (HA and Solarman cloud) has its benefits, I'm hoping to be able to run both.

david9600, 2025-04-13:
OK... I have a result from the packet traces.

HA is flooding the logger, with the result that the logger ends its TCP connection with the cloud, pauses for a couple of minutes, then re-opens the connection, only to run into the same trouble when it can't keep up with the TCP traffic HA is sending it.

I'm attaching a filtered trace output, which I've edited by replacing IP addresses with component names.  I've also highlighted the significant bits - orange for stumbles (retransmissions), red for TCP termination, and green for the subsequent TCP re-establishment.

A solution to the problem would be to reduce the TCP transmission window (the amount of data that can be sent before HA has to wait for an acknowledgement from the logger).  So, we need to work out how to do that (I'm not sure whether it needs to be done in the integration or HA itself).

[integ-enabled_logger-restarts-TCP.pdf](https://github.com/user-attachments/files/19723688/integ-enabled_logger-restarts-TCP.pdf)

Stormhand, 2025-04-13:
I suspected something like this. When the integration is active it's impossible to read/set the configuration in the cloud. It fails and starts counting 60 seconds to retry and fail again.

sofkaski, 2025-04-13:
In my setup I used to have Deye Cloud active parallel to HA and I experienced on HA side connection errors 3-5 times a day. Symptoms there were similar so that the LSW stick closed the connection and remained unresponsive to HA a couple of minutes. Eventually `ha-solarman` was able to reconnect. The stick was also unresponsive to its web interface during that time.

Some time ago I disabled Deye Cloud (by clearing its addresses from the stick configuration), but it did not have much impact. Maybe the frequency of errors have a little bit decreased. There is no clear pattern when the error occurs.

My judgement has been that the stick internally leaks resources in some situation and then eventually fails and after some time resets itself.

david9600, 2025-04-17:
FYI, I received a reply from SolarmanPV re confirmation of latest firmware (in case the issue was fixed by a newer version). They said the installed version LSW3_01_270A_SS_00_01.05.00.05 is already the latest.
So evidently the issue (instability of cloud connection when another entity is communicating with the logger) is either not yet officially reported, or is known but not yet fixed.  I won't follow up further just now, as I've decided to try direct communication via the inverter's RS-485 port, and will leave the WiFi stick to manage the cloud data.

Stormhand, 2025-04-17:
> FYI, I received a reply from SolarmanPV re confirmation of latest firmware (in case the issue was fixed by a newer version). They said the installed version LSW3_01_270A_SS_00_01.05.00.05 is already [...]

I will do the same. Created esphome device.

andreuka, 2025-04-24:
same here, following

davidrapan, 2025-05-04:
Hi guys, I am really very sorry that you are having such a issues. I'm running personally quite hefty number of devices but never seen anything like it. And is important to say that I almost always use modified versions of the code which is causing even higher utilization (by design) as a form of stress testing and thus be sure that default values are "safe" for most users.

So I for example know that 1 s polling does cause issues with the cloud described in this thread. 3 s is then cca the threshold where I again did not see any issues whatsoever... Even so I had tested many devices throughout the development it still can be per device issue.

However, I still think that it's rather caused by some individual nuances in your specific setups. Something must be slowing down HA <-> Logger communication. What does your Inverter Update Interval graph look like?

Experienced users could try tcpdump traffic of the internet<->inverter link then look what's going on. This will be proly the best way how to determine the issue.

david9600, 2025-05-05:
Thanks David ... yes if anyone else is able to run a packet trace on their setup, it would be interesting to know if it shows the same as mine, i.e. that after a rapid succession of requests from HA, the logger closes its cloud connection (then reopens after pausing, during which time the app shows no data).

Initially I thought maybe the TCP acknowledgment window was set too high, however the trace showed the logger was acknowledging each request, so that theory was ruled out.

Could you confirm whether your LSW-3 firmware version is the same as mine?
LSW3_01_270A_SS_00_01.05.00.05

I came to the conclusion that there are bug/s in the logger firmware.

Until we know the cause of the issue, I've left the logger managing the cloud data, and built an ESP-based RS485 interface as described by Sofar2MQTT, which shows promise.

andreuka, 2025-05-06:
My Module Version No： LSW3_32U_5406_1.0B
Extended System Version： V1.1.00.10

david9600, 2025-05-06:
OK ... that's quite a different numbering scheme to mine ... presumably there are different versions of firmware for the different makes/models of inverter supported by the LSW-3 (and maybe even depending on country of sale).  That would be enough to explain why some loggers work concurrently with HA & Solarman cloud, while others have difficulty.

davidrapan, 2025-05-06:
Mine are LSW3_32U_5406_1.06, it's the first time I see version like your's @david9600 though. 😆 Firmwares aren't generally tied to specific brands/models as logger is simply only providing the access or communication with inverters modbus. But there are most certainly some bad versions which do cause some (and not just communication quality) issues. Some with added encryption capabalities have even port 8899 closed or don't respond to discovery packets.

david9600, 2025-05-08:
The "LSW3....270A..." series of firmware seems to be for the Sofar ESI stackable system - that's what mine is, and a net search on those firmware idents finds other posts relating to the ESI series.  It uses the G3 external Modbus protocol, and was working with the integration using 'sofar_g3hyd.yaml', apart from the cloud disconnection issue.
Anyway, the MQTT solution is now up and running here (based on Sofar2mqtt, but with several mods to the code to suit the ESI hybrids).

davidrapan, 2025-05-08:
Firmwares will be most likely based on the logger's physical socket type. Your inverter uses G3 external Modbus protocol not the logger, that's simply not how it works but it's not important here. My best guess is that your firmware doesn't prioritize connection to the cloud so whenever integration hits with a request when logger uploads to the cloud it's interrupted (so higher update interval can help but don't have to eliminate the issue all together).

vvv2542, 2025-05-20:
Hi, I have Deye inverter and **two Deye batteries**. Every 5 minutes the logger sends information about itself, the inverter and each battery to DeyeCloud. When Solarman integration is running, information about the inverter is delayed in the cloud, and the battery information stops updating at all. The application shows batteries status offline. To partially solve the problem, I have to change TIMINGS_INTERVAL from 5 to 15 in const.py after each update. After that, all the information in the cloud appears, but sometimes with a slight delay. Please, make TIMINGS_INTERVAL configurable.

davidrapan, 2025-05-20:
Don't change `TIMINGS_INTERVAL`! Change `TIMINGS_INTERVAL_SCALE`.

trozman, 2025-06-26:
The same issue here. Deye hybrid inverter (SUN-12K-SG04LP3-EU) + Deye batteries.
After enabling solarman integration, batteries show as offline in Deye cloud app. Firstly one battery, then all 4.
After a while, settings for the inverter can not be updated via Deye cloud app.

I also vote for a configurable update interval setting via ui.

rtripon, 2025-07-13:
Hello,
Have a friend with Deye hybrid inverter 6K + Deye battery. 
His experience the same problem, when Solarman is enabled, he cant query inverter settings from the cloud.

vvv2542, 2025-08-13:
> A temporary workaround mentioned in some other ticket was to copy the definition file to `custom_folder`, for example `custom_components/solarman/inverter_definitions/custom/deye_p3.yaml` and change `update_interval`. Then on the Integrations page, update Inverter's configuration and point to this profile (`custom/deye_p3.yaml`). I already updated Solarman several times and it didn't revert the changes.

The definition file also changes periodically. With each update, this must be checked and, if necessary, the changes must be merged manually.

closed 2025-12-24 by davidrapan

ALERTua, 2026-11-12:
I can confirm that setting 15 seconds helps.

vvv2542, 2026-11-12:
> I can confirm that setting 15 seconds helps.

Try updating the firmware, starting with the logger. In my case, that helped, and everything is now running stably with a 5 second interval.