davidrapan/ha-solarman #519: LSW-3 can't connect

lukee12zakk, 2025-04-05:
### Description

I can't connect my loger with my solar setup. Just gives me errors.
Serial number begins with: 2393xxxxxx
firmware on the logger: LSW3_15_MQTT_270A_1.10
firmware on the inverter: unknown. Does not show
Also, the serial number, and model number are not shown.

### Reproduction Steps

1. install this using hacs
2. reboot home assistant and setup integration
3. connect loger stick and then gives error

### Log

2025-04-05 16:18:06.680 WARNING (SyncWorker_0) [annotatedyaml.constructors] YAML file /config/configuration.yaml contains duplicate key "themes". Check lines 5 and 6
2025-04-05 16:18:06.689 WARNING (SyncWorker_0) [annotatedyaml.constructors] YAML file /config/configuration.yaml contains duplicate key "sensor". Check lines 13 and 52
2025-04-05 16:18:06.751 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hass_agent which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.753 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration spook which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.755 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration fontawesome which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.756 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ui_lovelace_minimalist which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.757 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration pirateweather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.759 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration truenas which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.760 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration powercalc which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.761 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration llama_conversation which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.763 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dwains_dashboard which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.765 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ssh which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.766 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.768 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration spook_inverse which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.769 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration browser_mod which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.770 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ollama_conversation which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.772 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ble_monitor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.774 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.775 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration tplink_router which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:06.776 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration network_scanner which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-05 16:18:11.024 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ states("sensor.luke_s_bed_power_power") |float + states("sensor.luke_s_desk_power_power") | float   }}W) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2385, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 644, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2904, in _render_with_context
    return template.render(**kwargs)
           ~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2388, in forgiving_float_filter
    raise_no_default("float", value)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1942, in raise_no_default
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ states("sensor.luke_s_bed_power_power") |float + states("sensor.luke_s_desk_power_power") | float   }}W' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 761, in async_render_to_info
    render_info._result = self.async_render(  # noqa: SLF001
[... 144 lines cut ...]
2025-04-05 16:19:18.052 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x90 0x2 0x0 0x0
2025-04-05 16:19:18.053 DEBUG (MainThread) [pymodbus.logging] Frame check failed, possible garbage after frame, testing..
2025-04-05 16:19:18.053 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2025-04-05 16:19:18.053 DEBUG (MainThread) [custom_components.solarman.device] [2393929195] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 1 [FrameError: Invalid modbus response received]
[... cut ...]
2025-04-05 16:19:18.139 DEBUG (MainThread) [custom_components.solarman.device] [2393929195] Querying 03 ❘ 0x03 ~ 0000 - 0022 ❘ 0x0000 - 0x0016 #023 failed, attempts left: 0, aborting. [FrameError: Invalid modbus response received]
2025-04-05 16:19:18.139 INFO (MainThread) [custom_components.solarman.device] [2393929195] Closing connection to 192.168.1.80
2025-04-05 16:19:18.140 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-04-05 16:19:18.140 INFO (MainThread) [custom_components.solarman.device] [2393929195] Closing connection to 192.168.1.80

### Version

25.03.18

### Home Assistant Core Version

2025.4.1

### Home Assistant Operating System Version

15.1

davidrapan, 2025-04-06:
Hi, what brand and model this is?

lukee12zakk, 2025-04-06:
The inverter is a sofar solar hybrid inverter with a model number of HYD 3000-EP.

davidrapan, 2025-04-06:
You need to select appropriate profile (e.g. [zcs_azzurro-hyd-zss-hp.yaml](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/zcs_azzurro-hyd-zss-hp.yaml)). Autodetection is only for Deye compatible devices.

lukee12zakk, 2025-04-06:
Thanks. That fixed it. I've never done this before and this was all very confusing for me, so thank you for helping me.

closed 2025-04-06 by lukee12zakk