davidrapan/ha-solarman #528: ModbusProxy - connection to modbusproxy successfull, reset by the peer

themmerle, 2025-04-10:
### Description

Has anyone this integration running successfully over a modbusproxy? I have evcc running perfectly over the proxy (so the proxy must be configured correctly). When I specify the IP of the logger directly, it works. If I specify the IP of the proxy, it does not work. Port and Network settings should be fine, otherwise evcc would not succeed with Inverter, Meter, Batterie.

Thanks a lot for your help.

### Reproduction Steps

[solarman_log.txt](https://github.com/user-attachments/files/19686781/solarman_log.txt)

### Log

2025-04-10 16:29:50.312 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Successful reconnection! Data expected. Will retry the last request
2025-04-10 16:29:50.313 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Connection is reset by the peer. Will try to restart the connection
2025-04-10 16:29:50.315 INFO (MainThread) [custom_components.solarman.device] [2107826015] Closing connection to 192.168.178.135
2025-04-10 16:29:50.316 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 20.002 seconds (success: False)
2025-04-10 16:29:50.316 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2025-04-10 16:29:50.316 INFO (MainThread) [custom_components.solarman.device] [2107826015] Closing connection to 192.168.178.135
2025-04-10 16:29:55.377 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2025-04-10T11:56:30.475334+00:00', 'data': {'name': 'Inverter'}, 'discovery_keys': {}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01JRFRY2WBKZDYXXRMC056HGVC', 'minor_version': 7, 'modified_at': '2025-04-10T14:18:05.363821+00:00', 'options': {'host': '192.168.178.135', 'lookup_file': 'zcs_azzurro-hyd-zss-hp.yaml', 'sn': 2107826015}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'subentries': [], 'title': 'Inverter', 'unique_id': None, 'version': 1})
2025-04-10 16:29:55.378 DEBUG (MainThread) [custom_components.solarman] async_setup_entry: config_entry.runtime_data.async_config_entry_first_refresh
2025-04-10 16:29:55.378 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: Broadcasting on 192.168.178.135
2025-04-10 16:29:55.380 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to 192.168.178.135
2025-04-10 16:29:55.884 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on [IPv4Network('192.168.178.0/24'), IPv4Network('172.30.32.0/23'), IPv4Network('172.30.232.0/23')]
2025-04-10 16:29:55.886 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-04-10 16:29:55.887 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Send to ['192.168.178.255', '172.30.33.255', '172.30.233.255']
2025-04-10 16:29:55.889 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: [192.168.178.143, E8FDF88B541D, 2107826015]
2025-04-10 16:29:56.393 DEBUG (MainThread) [custom_components.solarman.provider] Device autodetection is disabled. Selected profile: zcs_azzurro-hyd-zss-hp.yaml
2025-04-10 16:29:56.393 DEBUG (MainThread) [custom_components.solarman.discovery] DiscoveryProtocol: Connection closed
2025-04-10 16:29:56.647 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 60, code: 3, min_span: 25, max_size: 125, digits: 6, attributes: {'mod': 0, 'mppt': 4, 'l': 3, 'pack': -1}
2025-04-10 16:29:56.651 DEBUG (MainThread) [custom_components.solarman.device] {2107826015: {'connections': {('mac', 'e8:fd:f8:8b:54:1d')}, 'identifiers': {('solarman', '2107826015')}, 'configuration_url': 'http://192.168.178.143/config_hide.html', 'serial_number': '2107826015', 'manufacturer': 'Zcs', 'model': 'AZZURRO-HYD-ZSS-HP', 'name': 'Inverter'}}
2025-04-10 16:29:56.652 DEBUG (MainThread) [custom_components.solarman.device] [2107826015] Scheduling 9 query requests. ^0
2025-04-10 16:29:56.653 DEBUG (MainThread) [custom_components.solarman.device] [2107826015] Querying 03 ❘ 0x03 ~ 1028 - 1067 ❘ 0x0404 - 0x042B #040 ...
2025-04-10 16:29:56.657 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Successful connection!
2025-04-10 16:29:56.657 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] SENT: a5 17 00 10 45 31 00 5f df a2 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 5a 15
2025-04-10 16:29:56.659 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Connection is reset by the peer. Will try to restart the connection
2025-04-10 16:29:56.660 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Successful reconnection! Data expected. Will retry the last request
2025-04-10 16:29:56.662 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Connection is reset by the peer. Will try to restart the connection
2025-04-10 16:29:56.664 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Successful reconnection! Data expected. Will retry the last request
2025-04-10 16:29:56.665 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Connection is reset by the peer. Will try to restart the connection
2025-04-10 16:29:56.667 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [2107826015] Successful reconnection! Data expected. Will retry the last request

### Version

25.02.18

### Home Assistant Core Version

2025.4.1

### Home Assistant Operating System Version

14.2

themmerle, 2025-04-10:
LSE-3 Firmware versionME_0D_270A_1.08

81ondo, 2025-04-10:
@themmerle are you suggesting me to update the firmware of the inverter?

themmerle, 2025-04-11:
sorry, my mistake! I was mixing up the firmware logger / inverter. Just wanted to say that I am using a LSE-3 logger. Firmware Version unknown ;-)

davidrapan, 2025-05-04:
Hi @themmerle, set serial number to 0 in device configuration.

closed 2025-06-22 by github-actions[bot]