davidrapan/ha-solarman #531: Sofar ME3000SP changing country the whole time

peterdedecker83, 2025-04-16:
### Description

Hi,

See this log. I guess this is not normal behaviour and might be a bug. Can we verify this?

[image]

### Reproduction Steps

Sofar ME3000SP just in production, see log

### Log

[home-assistant_solarman_2025-04-16T21-37-01.118Z.log](https://github.com/user-attachments/files/19785073/home-assistant_solarman_2025-04-16T21-37-01.118Z.log)

### Version

25.03.18

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.1

davidrapan, 2025-05-04:
Hi @peterdedecker83, it's most likely caused by incorrect register in the entity description. What profile it is?

peterdedecker83, 2025-05-04:
Hi @davidrapan I'm using the sofar_wifikit.yaml profile.

davidrapan, 2025-05-05:
So the profile seems to match available documentation. No errors there. What country is set in the device really? Are you sure it's not changing for who know what reason there too?

davidrapan, 2025-05-05:
I mean, when it changes to Germany did you try to check what value is in the inverter settings? Also in the log you uploaded is every reading with value 8: Belgium so try to record longer one.

But still, I don't think this is something we can solve here..

peterdedecker83, 2025-05-05:
How can I check what the value in the inverter settings is? Only at the device itself? That seems impossible, as it is always for a very short time (max 10s) that the value returned with solarman is Germany, it's Belgium during more than 99.9% of the time.

[image]

I've added a very long log in attach which should include these values.

[home-assistant_solarman_2025-05-05T20-56-11.968Z.log](https://github.com/user-attachments/files/20048691/home-assistant_solarman_2025-05-05T20-56-11.968Z.log)

davidrapan, 2025-05-06:
> How can I check what the value in the inverter settings is? Only at the device itself? That seems impossible, as it is always for a very short time (max 10s) that the value returned with solarman is Germany, it's Belgium during more than 99.9% of the time.

That's very short time for it, yes.

In the log is visible that sometimes arrives response with it' second half filled just with zeroes which means that it affects more entities and not just Country.

To be completely sure that the issue doesn't come from the integration try to downgrade to v24.12.22 and see if it changes anything.

closed 2025-07-12 by github-actions[bot]