davidrapan/ha-solarman #533: Max PV Power max too low

haysdb, 2025-04-17:
### Description

I have a Deye 10 kW hybrid inverter. Max AC output is 10,000 watts but max PV DC input is 13,000. DC-connected batteries can use power above the AC limit but integration sets a maximum PV Max Power of 12,000. I can set it to anything I want on the inverter. This is an edge case, but today I saw Solar Irradiance of nearly 1,200 w/m2 and PV power briefly above 13,000 for my 12.71 kW solar array. So my request is to increase the maximum for this entity to match the datatsheet for the inverter of 13,000.

### Reproduction Steps

Easy. You just need 1200 w/m2 of solar radiation. :-)

### Log

[image]

### Version

25.03.18

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.2

haysdb, 2025-04-17:
I fixed this myself by editing the file deye_hybrid.yaml

davidrapan, 2025-05-04:
[fix: PV Max Power range in Deye](https://github.com/davidrapan/ha-solarman/commit/b32ca8cc9987535251b8ea587c42b56b8ca16536)

closed 2025-05-04 by davidrapan