davidrapan/ha-solarman #534: Device fault DC over voltage always shown.

networxnet, 2025-04-17:
### Description

Integration always shows an Device Fault "DC Bus Over voltage", while inverters web console shows no error/alert.

Wondering 'cause only one panel is connected delivering about 60W on this rainy day.

### Reproduction Steps

Restart of inverter, reconfiguration of integration

### Log

`00 01 00 08 00 01 00 01 00 00 00 01 ff 54 00 02 00 0c 27 10 00 00 27 10 00 00 00 00 00 00 00 02 00 01 00 00 00 00 04 ae 00 00 00 01 00 00 00 00 00 00 02 47 00 00 00 00 00 00 09 4c 00 e6 00 00 00 02 01 57 00 00 13 7d 00 00 00 00 00 00 00 00 00 00 00 00 01 ee 00 00 00 00 00 00 0a 00 09 84 09 84 13 7f 00 00 00 00 00 00 00 00 00 00 00 00 02 ab 00 02 00 1d 00 02 00 00 00 00 00 00 00 00 00 00 01 54 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 2b 7d 92 15
2025-04-17 15:18:11.785 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0xc0 0x0 0x41 0x19 0x4 0x11 0xf 0xc 0x3b 0x1 0xb0 0x0 0xc 0xb 0x3b 0x7 0x30 0x14 0x1e 0x12 0x8e 0x9 0xe2 0x4 0xb 0x0 0x1 0x13 0x9c 0x0 0x28 0x13 0x74 0x13 0x9c 0x0 0x64 0x7 0xd0 0x0 0x64 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x1 0x0 0x8 0x0 0x1 0x0 0x1 0x0 0x0 0x0 0x1 0xff 0x54 0x0 0x2 0x0 0xc 0x27 0x10 0x0 0x0 0x27 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x1 0x0 0x0 0x0 0x0 0x4 0xae 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x47 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x4c 0x0 0xe6 0x0 0x0 0x0 0x2 0x1 0x57 0x0 0x0 0x13 0x7d 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xee 0x0 0x0 0x0 0x0 0x0 0x0 0xa 0x0 0x9 0x84 0x9 0x84 0x13 0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xab 0x0 0x2 0x0 0x1d 0x0 0x2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x54 0x0 0xf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2b 0x7d
2025-04-17 15:18:11.785 DEBUG (MainThread) [pymodbus.logging] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[65, 6404, 4367, 3131, 432, 12, 2875, 1840, 5150, 4750, 2530, 1035, 1, 5020, 40, 4980, 5020, 100, 2000, 100, 0, 0, 1, 1, 8, 1, 1, 0, 1, 65364, 2, 12, 10000, 0, 10000, 0, 0, 0, 2, 1, 0, 0, 1198, 0, 1, 0, 0, 0, 583, 0, 0, 0, 2380, 230, 0, 2, 343, 0, 4989, 0, 0, 0, 0, 0, 0, 494, 0, 0, 0, 2560, 2436, 2436, 4991, 0, 0, 0, 0, 0, 0, 683, 2, 29, 2, 0, 0, 0, 0, 0, 340, 15, 0, 0, 0, 0, 0, 0], status=1) 
2025-04-17 15:18:11.786 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2025-04-17 15:18:11.786 DEBUG (MainThread) [custom_components.solarman.device] [3847504281] Querying 03 ❘ 0x03 ~ 0021 - 0116 ❘ 0x0015 - 0x0074 #096 succeeded
2025-04-17 15:18:11.787 DEBUG (MainThread) [custom_components.solarman.parser] Value: 0.0 of total_production_2_sensor validation failed: {'min': 0.1}
2025-04-17 15:18:11.787 DEBUG (MainThread) [custom_components.solarman.device] [3847504281] Returning 49 new values. [Previous State: Connected (1)]
2025-04-17 15:18:11.787 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter Deye K23 data in 0.619 seconds (success: True)
2025-04-17 15:18:16.171 DEBUG (MainThread) [custom_components.solarman.device] [3847504281] Scheduling 1 query request. ^670
2025-04-17 15:18:16.171 DEBUG (MainThread) [custom_components.solarman.device] [3847504281] Querying 03 ❘ 0x03 ~ 0021 - 0116 ❘ 0x0015 - 0x0074 #096 ...`

### Version

25.03.18

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.2

CrazyUs3r, 2025-04-18:
That's a problem I have with my Sun M80 and Sun 800. But for you with a panel, that's normal.

networxnet, 2025-04-18:
> That's a problem I have with my Sun M80 and Sun 800. But for you with a panel, that's normal.

Why this should be notmal

networxnet, 2025-04-18:
Why this should be normal?

On the one hand, I see no reason for this error, as all parameters are within the target range.

On the other hand, there is a discrepancy between what the inverter displays in its web console (no error/alarm) and what the integration displays (error).

networxnet, 2025-04-18:
> That's a problem I have with my Sun M80 and Sun 800. But for you with a panel, that's normal.

Why this should be normal?

On the one hand, I see no reason for this error, as all parameters are within the target range.

On the other hand, there is a discrepancy between what the inverter displays in its web console (no error/alarm) and what the integration displays (error).

> That's a problem I have with my Sun M80 and Sun 800. But for you with a panel, that's normal.

CrazyUs3r, 2025-04-18:
I don't think deye will put any more effort into the web interface. Bugs will only be reported in the deye cloud app or solarman app.

davidrapan, 2025-05-04:
Hi @networxnet, are all the MPPT inputs connected/utilized?

ulope, 2026-08-24:
And yet the official apps don't show an error