davidrapan/ha-solarman #535: sensor.inverter_total_energy from integration solarman has state class total_increasing, but its state is not strictly increasing

768g, 2025-04-18:
### Description

Entity sensor.inverter_total_energy from integration solarman has state class total_increasing, but its state is not strictly increasing. Triggered by state 4879.4 (4879.6) with last_updated set to 2025-04-18T11:58:20.842292+00:00. Please create a bug report at https://github.com/davidrapan/ha-solarman/issues

### Reproduction Steps

Install HA-Solarman and wait for aggregation

### Log

`Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:366
integration: Sensor (documentation, issues)
First occurred: 2:00:10 PM (1 occurrences)
Last logged: 2:00:10 PM

Entity sensor.inverter_total_energy from integration solarman has state class total_increasing, but its state is not strictly increasing. Triggered by state 4879.4 (4879.6) with last_updated set to 2025-04-18T11:58:20.842292+00:00. Please create a bug report at https://github.com/davidrapan/ha-solarman/issues`

### Version

25.03.18

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.2

CrazyUs3r, 2025-04-18:
Witch inverter with witch yaml you use, and Post a debug log when you can Catch it

davidrapan, 2025-05-04:
Hi @768g, I have honestly no idea what could possibly be the cause of this fluke. Is this an isolated case? Or does it happen regularly?