davidrapan/ha-solarman #536: Get Off Grid

Mights001, 2025-04-19:
### Describe the solution you'd like

Hi David can you add this Register to get Offgrid for the LV Invertes and i think it's the same for the HV
[image]

### Describe alternatives you've considered

no Matter :)

strezitom, 2025-04-25:
Nice Trick!
I came across a register that forces the inverter Off grid, but that did not work. Reg 179, Force Offgrid

Will give this a go, but I see no reason why this would not work :D
12kW SG04LP3-EU

strezitom, 2025-04-25:
I can confirm, this works like a treat! Thanks @Mights001 for sharing!

I added this to deye_p3.yaml:
      - name: "Grid Over Freq Trip 2"
        platform: number
        class: "frequency"
        state_class: "measurement"
        uom: "Hz"
        scale: 0.01
        rule: 1
        registers: [0x0167]
        configurable:
          min: 47
          max: 51.5
          step: 0.5
        range:
          min: 4700
          max: 5150

DavidKarlas, 2025-07-29:
Do you guys think forcing OffGrid via 179 will make house appliances and inverter safe from indirect lighting strike?