davidrapan/ha-solarman #537: Deye hybrid Microinverter control

jdpmendes, 2025-04-19:
### Is your feature request related to a problem?

Hello,

I have a DEYE Sg05lp3 (registers as an sg04lp3), where I use the Gen port for microinverters.
The inverter has the option to cut off export to grid (def Disabled), and two addional SOC percentages OFF (def 100%) and ON (def 95%).

### Describe the solution you'd like

These 3 settings are currently missing in the integration. Would it be possible to add them?

Thanks

### Describe alternatives you've considered

None

jdpmendes, 2025-04-21:
[image]

jdpmendes, 2025-04-21:
[image]

kris66dfg, 2025-04-24:
Hello
I urgently need to control this option, can you add it to the integration??
"MI export  to Grid  cut Off

because I see that it is missing

cut off export to grid

davidrapan, 2025-05-04:
Hi @jdpmendes, the relevant entity has been added already into the code and thus will be part of the next release.

If you don't want to wait you can manually add Microinverter Export cut-off entity from [deye_hybrid.yaml](https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/inverter_definitions/deye_hybrid.yaml) to the file in HA at `/config/custom_components/solarman/inverter_definitions/deye_hybrid.yaml`

davidrapan, 2025-05-09:
[feat: Add MI Export cut-off and Gen/Grid Peak shaving for Deye](https://github.com/davidrapan/ha-solarman/commit/169e9a6bc733cf2dbc75d5b85ae0ed4c16352a2a)

closed 2025-05-09 by davidrapan