davidrapan/ha-solarman #538: Sofar ME3000SP cannot set (dis)charge power

peterdedecker83, 2025-04-21:
### Description

Setting the (dis)charging speed using the solarman.write_holding_register functions to address 258 or writing 0 to 259 to return to auto doesn't work.

### Reproduction Steps

Use one out of the three different functions to write to a (holding) register on the Sofar device, write 1500 to address 258 or 0 to 259

### Log

[home-assistant_solarman_2025-04-21T20-16-45.817Z-charging-speed.log.txt](https://github.com/user-attachments/files/19839920/home-assistant_solarman_2025-04-21T20-16-45.817Z-charging-speed.log.txt)

### Version

25.03.18

### Home Assistant Core Version

 2025.4.2 

### Home Assistant Operating System Version

15.2

peterdedecker83, 2025-04-22:
Same problem using the new registers 304 and following

[home-assistant_solarman_2025-04-22T07-55-39.131Z.log](https://github.com/user-attachments/files/19845781/home-assistant_solarman_2025-04-22T07-55-39.131Z.log)

davidrapan, 2025-05-04:
Hi @peterdedecker83, device proly requires setting all the relevant registers to (dis)charge in bulk. That means you are currently forced to "fallback" using built-in services for writing registers.

stefan088, 2025-05-05:
since last update, also register 4608 (% , auto) doesn't work anymore

davidrapan, 2025-05-10:
Hi @stefan088, what do you mean?

stefan088, 2025-05-11:
> Hi [@stefan088](https://github.com/stefan088), what do you mean?

fixed with latest update, probably related to issue #570