davidrapan/ha-solarman #542: daily sensor

italoc, 2025-04-25:
### Is your feature request related to a problem?

your integration is great but i can't find this sensor

Daily Battery Charge
Daily Battery Discharge
Daily Energy Bought
Daily Energy Sold
Daily Load Consumption
Daily Production

### Describe the solution you'd like

in this integration is present: https://github.com/StephanJoubert/home_assistant_solarman

### Describe alternatives you've considered

in alternative you can create this sensor with utility meter

rtripon, 2025-04-26:
Hello,
Those exists for sure, because I have them all, and I didnt do anything specific to have them.

[image]

italoc, 2025-04-26:
sorry, my fault

closed 2025-05-04 by davidrapan