davidrapan/ha-solarman #547: Total Power Grid and External CT Power L1L2 has no entity

ccfiel, 2025-04-30:
### Description

Total Power Grid and External CT Power L1L2 have no registered entity. Do you have any idea how to fix this issue? 

Inverter is Deye Hybrid

[image]

### Reproduction Steps

[image]

### Log

No logs

### Version

25.03.18

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

20250411.0

PV-Joe, 2025-05-01:
I have a Deye Hybrid and the entities are there.

Delete the filter before searching and they probably show up?

[image]

ccfiel, 2025-05-02:
@PV-Joe  still the same? :( 

[image]

ccfiel, 2025-05-02:
Are we the same configuration? @PV-Joe 
[image]

davidrapan, 2025-05-04:
Hi @ccfiel, try to remove and re-add the device back again.