davidrapan/ha-solarman #548: Peak shaving value range lower than actually available in inverter for Deye SUN-12K-SG04LP3

toruonu, 2025-05-01:
### Description

I run Solarman 25.03.18 and when I try to set peak shaving value it lists it as a valid range of 0 - 16 000. However the inverter can AC pass through a lot more and on the inverter screen I was able to set it to my grid connection capacity of 22 000. It works, it does peak shaving there, but I cannot tune the value from solarman. As it's pretty much fixed I don't care much, but it also doesn't display the 22 000 value in Solarman so decided to report it. 

### Reproduction Steps

Try to add a value higher than 16 000 to the peak shaving variable and you get notification that it is out of range. Works on inverter though. 

### Log

As this is fixed range limitation do you need the log?

### Version

25.03.18

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

nan

toruonu, 2025-05-01:
Also, I can't seem to turn peak shaving on-off. Don't find the respective true-false flag.

davidrapan, 2025-05-07:
[fix: Peak shaving range in Deye](https://github.com/davidrapan/ha-solarman/commit/c2534d615a11d8e51f8f9e149852b025c0661ee2)

closed 2025-05-07 by davidrapan