davidrapan/ha-solarman #549: problem to connect invt_xd-tl

superprp, 2025-05-02:
hello, i have download with HACS this integration and when i try to add it its required a SN (from inverter or from logger?) i have tried both (first one have letters too so i think is the logger sn but it doesnt work, i have left host or ip adress blank and port 8899 (by default), any suggestion? thanks

davidrapan, 2025-05-04:
Hi @superprp, all the necessary connection parameters should be in most cases prefilled automatically through device discovery (when is inverter and HA on the same network). So if that didn't happen there is proly something wrong on your side.

Possible culprits:
* Device is unreachable (from the HA instance): Try ping logger's IP directly from HA
* Incompatible Logger (alphanumeric serial number): Can be checked on the device itself
* Logger is running firmware with port 8899 disabled: You can check open ports on the logger's IP using **nmap**
  * The only solution in this last case is to ask solarman to downgrade logger's firmware.

superprp, 2025-05-04:
The problem i have js that the company Who install the inverter installed a router preconfigured with the logger, and i dont have access yo this router, this router is connected to my main router and i can connect to the logger thro solarman app
Do you know could i get the IP of the logger?
My logger is model lsw-5, i dont know if It IS compatible or not

davidrapan, 2025-05-04:
So the logger is not connected to your network. The only option is then ask the company to forward port tcp 8899 in the router to which is logger connect and forward this port to loggers IP. Then you will be able to use that routers IP in device configuration.