davidrapan/ha-solarman #551: Failed setup - Device setup timed out

MajorNR, 2025-05-05:
### Description

Hi,after I installed integration using HACS, restarted HA and add all informations ( ID of data logger, IP of logger discovered on LAN, and deye_P3 file ) and right after this problem occured Failed setup [LOGGERID] - Device setup timed out. I cant get any info from the logger, devices nor entities are loaded into HA. 

[image]

[image]

### Reproduction Steps

- installed from HACS
- restarted HA
- inputed all requested infos
- integration, device and entities are not working
- integrations is still initialising

### Log

[home-assistant_solarman_2025-05-05T17-06-47.996Z.log](https://github.com/user-attachments/files/20040836/home-assistant_solarman_2025-05-05T17-06-47.996Z.log)

### Version

25.03.18

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-05:
Hi @MajorNR, can you ping the logger directly from HA? Does the logger's serial number contains only numbers? What firmware version the logger has? Some of the newest version has port 8899 closed.

davidrapan, 2025-05-05:
Debug log? And Force reload the integration during logging.

amiko24, 2025-05-05:
[home-assistant_solarman_2025-05-05T19-59-20.791Z.log](https://github.com/user-attachments/files/20047744/home-assistant_solarman_2025-05-05T19-59-20.791Z.log)

davidrapan, 2025-05-05:
Even so the log doesn't contains what I asked for I can see that the communication is happening just fine so what exactly is the issue @amiko24?

amiko24, 2025-05-05:
Today I had sit 13:30 to approx. 21:30 problems with the connection to the inverter. 

[image]

After 9pm I restarted FritzBox and Intel NUC with HA and the connection has been working since then.

Thank you, I will look at the connection soon.

MajorNR, 2025-05-06:
> Hi [@MajorNR](https://github.com/MajorNR), can you ping the logger directly from HA? Does the logger's serial number contains only numbers? What firmware version the logger has? Some of the newest version has port 8899 closed.

Hi, thank you for your quick reply, please find answers below :
- can you ping the logger directly from HA? No, pinging using ICMP integrations shows device as disconnected. But I can connect to it using same IP adress via search bar in browser. 
- Does the logger's serial number contains only numbers? Yes only numbers starting with 3, displayed as Device serial number
- What firmware version the logger has? LSW3_32U_5406_1.0B
- Some of the newest version has port 8899 closed - how to check this?

davidrapan, 2025-05-06:
You can check open open ports on specific IP address using nmap, but according to your ping test your logger is not reachable from HA. You need to solve that first. What is your network layout? Are you sure both HA and logger are members of the same network?

amiko24, 2025-05-06:
> It looks more like an issue in your network, or logger? I'm on LSW3_32U_5406_1.06. Haven't you maybe received an update recently?

I cannot say whether I have received an update. I have 2 systems both with the Deye inverter the loggers are the same, on the second system there is no problem with the communication.  Maybe the problem is with the FritzBox, I'll keep an eye on it.

MajorNR, 2025-05-07:
> You can check open open ports on specific IP address using nmap, but according to your ping test your logger is not reachable from HA. You need to solve that first. What is your network layout? Are you sure both HA and logger are members of the same network?

I realised that they wer not in same network. It is working now, thank you :)

davidrapan, 2025-05-07:
> I realised that they wer not in same network. It is working now, thank you :)

It happens. :)

@amiko24, if the problem returns then open a new issue. 😉

closed 2025-05-07 by davidrapan