davidrapan/ha-solarman #553: Sofar HYD 10KTL-3PH - Unstable connection

jcz1, 2025-05-08:
### Description

Yesterday i migrated Solarman integration from StephanJoubert to this (thanks!). Unfortunately i run into connection unstability. Many small disconnects and then one big - i had to restart HA. IP of FVE stick was unavailable in that moment (ICMP ping).
Also tried integration downgrade to version 24.12.22. It's much better (screen 2) but not rock stable. Maybe it is because too aggressive interval?

I have latest HAOS and all addons except solarman now.
My invertor: HYD 10KTL-3PH firmware 0V000001
My stick: LSW3 (wifi) firmware 15_270A_1.68
Invertor is in the same small room with wifi AP.

[image]

[image]

### Reproduction Steps

Install latest integration and connect invertor. There are data for some seconds but cycling with N/A.

### Log

```
2025-05-07 22:48:47.525 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching Sofar data
```

### Version

25.03.18

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-08:
Hi @jcz1, yeah there are issues in the latest versions caused by pymodbus library. I'm already working on migration back to umodbus (used in v24.12.22).

Try to increase update interval directly in used profile and see if anything changes. 😉 But! Single disconnect now and then throughout few hours is alright in my opinion if the disconnects don't take too long and sensors keeps their value.

davidrapan, 2025-05-09:
Enable debug logging too so I can check if it's simple timeout every time.

kkmaslowski, 2025-05-10:
I have the same problem with LSW3. 

[image]

davidrapan, 2025-05-10:
Hi @kkmaslowski, can you also record some debug log for me too?

davidrapan, 2025-05-12:
Hey @jcz1, seen the log, but apart from the disconnections, there's nothing too unusual (like other errors w/ requests or something). But I was looking for some information about the firmware version of your logger and found several complaints about the connection and some were able to resolve it by updating to 1.69. So I think you should try asking Solarman for an update.. or I can send you a download link from the official server.

jcz1, 2025-05-12:
@davidrapan thanks for informations. If i can update stick firmware and it is reasonably safe, please send me the link.

Meanwhile i upgraded HA and integration to the latest versions (v25.05.10). Connection losses again, sent new debug logs.

davidrapan, 2025-05-12:
Some entities had incorrect units of measurement, which have been fixed in the latest version.

jcz1, 2025-05-13:
stick fw 1.69, integration 25.05.12, even with default update_interval: 60

[image]

Maybe there are too much entities? I will try to reduce them.

davidrapan, 2025-05-13:
I'm 90% sure it's not about the number of entities... Did it work like this with the default profile or did you have a modified one? Because in the last release I adjusted the intervals a bit so that there were as few queries as possible. Can you upload a log for me again, even a longer one with the default profile? And you can also try 1.70 and 1.71 because there's nothing to lose.

jcz1, 2025-05-13:
Ok i will. That was default sofar_g3hyd profile from 25.05.12. Later i modified one parameter only (default interval).

davidrapan, 2025-05-13:
Wha you can also try is to increase timeout to 30s. In const.py at line 98: change 4 to 6.

jcz1, 2025-05-13:
Latest defaults. Debug logs uploaded.

[image]

davidrapan, 2025-05-13:
It really looks like some very bad WiFi or connection. Almost every time the scheduled requests take 20 seconds to transfer. Very strange, but I see one small problem with the integration. The timeout should be from the last response from the logger and not from the start of the transfer. Currently it cuts many connections that would be successful.

kkmaslowski, 2025-05-14:
LSW3 have COM serial port, maybe that's the reason of slow responses?

kkmaslowski, 2025-05-14:
@davidrapan does your stick also send data to cloud?

davidrapan, 2025-05-14:
Yes, because I need to know if cloud features works alongside using this integration.

davidrapan, 2025-05-15:
Hey @jcz1, can you try something? In `/config/custom_components/solarman/device.py`:
* Change line 103 so it looks like this:
```
            async with asyncio.timeout(TIMINGS_UPDATE_TIMEOUT) as timeout:
```
* And on line 109 add this:
```
                        timeout.reschedule(TIMINGS_UPDATE_TIMEOUT)
```

jcz1, 2025-05-15:
I've been experimenting with wifi. I have a Unifi AP and indeed both sticks (Sofar inverter, Pylontech battery) pretend to have a bad "experience", some TX retries and some disconnects (not quite corresponding to the disconnections in HA). They are about 1,5m away from the AP and good signal (-40 or -50dBm). I have tried different configurations include IoT compatibilities and it didn't improve much.

Our sticks are not connected to the cloud and on firewall there is IP reject to internet too. I know there are some problems with wifi on SoC with 1 CPU core like older ESP so some process is able to block wifi. Maybe something like this?

Anyway I haven't seen any problems with disconnections with old integration from StephanJoubert but I don't know how much he was evaluating error states or timeouts.

Testing with edited device.py now.

jcz1, 2025-05-15:
> Testing with edited device.py now.

There is some error with integration settings now. Debug log uploaded. Code diff:

```
➜  solarman diff device.py.zal device.py
103c103
<             async with asyncio.timeout(TIMINGS_UPDATE_TIMEOUT):
---
>             async with asyncio.timeout(TIMINGS_UPDATE_TIMEOUT) as timeout:
108a109
>                         timeout.reschedule(TIMINGS_UPDATE_TIMEOUT)
```

davidrapan, 2025-05-15:
Ah the other line has to be like this:
```
                         timeout.reschedule(asyncio.get_running_loop().time() + TIMINGS_UPDATE_TIMEOUT)
```

jcz1, 2025-05-16:
> Oh, the second line must be like this:
> 
> ```
>                          timeout.reschedule(asyncio.get_running_loop().time() + TIMINGS_UPDATE_TIMEOUT)
> ```

Rock stable! 🎉 

[image]

davidrapan, 2025-05-16:
Great, but show me the Update Interval graph.

jcz1, 2025-05-16:
Connected for 30+ hours now. Graph here:

[image]

I will try default profile (10s interval, i have 60) and default const.py too (timeout 4 vs 6).

jcz1, 2025-05-16:
[image]

davidrapan, 2025-05-16:
It's crazy how it sometimes takes almost a minute to complete all the requests... So I wonder what's slowing it down so much. The loggers are nothing special in terms of hardware and they also have very poorly written firmware, but we are talking about such a small amount of transmitted data that it is really a out of the ordinary, weird case, hah (assuming your network is completely fine).

jcz1, 2025-05-23:
This week include yesterday's update to latest integration. All default.
My plan is test newer stick firmware because there are still errors on the network level.

[image]

[image]

davidrapan, 2025-05-23:
It is really difficult to investigate network-related issues on standard network settings. If you were running OpenWrt as your router, we could look at the traffic between the HA and the inverter.

davidrapan, 2025-05-27:
Loggers are constantly reconnecting to wifi, for some reason, so maybe the errors comes from that.

davidrapan, 2025-05-29:
> Still don't know why there are these wifi reconnects.

As I said. The firmware is nuts. When I look at the wifi logs, it looks periodic and intentional or something.

But I'm glad that it's (at least somewhat) working. 😉

closed 2025-06-01 by davidrapan

zulufoxtrot, 2025-08-03:
Hi @davidrapan,

> I can send you a download link from the official server.

Would you mind sharing this?

FYI, my current version is LSW3_15_FFFF_1.0.65. The S/N of my device starts with 23.

Thanks!