davidrapan/ha-solarman #558: Sofar LSW3 Unable to pull valid data

Saad-Imtiaz, 2025-05-09:
### Description

Hello,

I have a 15kW Sofar grid tied inverter along with the Sofar LSW3 Logger attached. The integration is unable to pull valid data from the logger. 
If I use the auto configuration it selects the wrong config and shows no data, and if I select the LSW3 configuration I don't get any valid data, while on the solarman app everything is shown correctly. 

I am attaching the screenshots for reference, I hope it helps. 

Thank you for making this integration and thank you in advance for the help. 

KR

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Reproduction Steps

1. Install the integration 
2. Configure - Add the IP Address of the Inverter.
3. Select the LSW3 configuration

### Log

[home-assistant_solarman_2025-05-09T00-52-44.261Z.log](https://github.com/user-attachments/files/20113397/home-assistant_solarman_2025-05-09T00-52-44.261Z.log)

### Version

25.05.08

### Home Assistant Core Version

2025.4.2

### Home Assistant Operating System Version

15.1

davidrapan, 2025-05-09:
Hi @Saad-Imtiaz, what exactly is the inverter model?

Saad-Imtiaz, 2025-05-09:
Hi @davidrapan , The exact model of the inverter is SOFAR 15KTLX-G3.
And yes I tried the other configs, they didn't work.

davidrapan, 2025-05-09:
I think you should be using `sofar_tlx-g3.yaml` but there is currently an error which was corrected in [fix: IllegalDataAddressError w/ sofar-tlx-g3 profile](https://github.com/davidrapan/ha-solarman/commit/35db4accc95af1a5df0f60a0d96103c7c9944999)

Saad-Imtiaz, 2025-05-10:
Hi @davidrapan , 

I have tired all of them, and I get the error of 'CRC validation failed' on all of them except the LSW3.

[image]

[image]

davidrapan, 2025-05-10:
Update the integration to [v25.05.10](https://github.com/davidrapan/ha-solarman/releases/tag/v25.05.10)

Saad-Imtiaz, 2025-05-11:
Tried it just now, same problem.

Has the same CRC validation failed error on the Sofar TLX config, and the LSW3 is not showing correct data.

davidrapan, 2025-05-11:
Can you please enable debug logging and attach the file here? Thank you.

davidrapan, 2025-05-13:
Duplicate of #142

Saad-Imtiaz, 2025-05-13:
Oh Thanks, I can give that a try, hope it gets fixed.... 
I was actually thinking of making some hardware to get the data from RS485 Port and use an ESP32 to send it to home assistant... I'll update here how it goes.

davidrapan, 2025-05-13:
Great! 👍

closed 2025-05-13 by davidrapan