davidrapan/ha-solarman #560: Error in helpers/entity_platform.py:382 after update to 25.05.08

andreuka, 2025-05-09:
### Description

Updated to latest version today and solarman does not starts anymore

### Reproduction Steps

Update to 25.05.08, solarman does not starts anymore

### Log

Error while setting up solarman platform for sensor: unsupported operand type(s) for |: 'dict' and 'set'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/solarman/sensor.py", line 56, in async_setup_entry
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, config_entry.options), d) for d in descriptions)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 551, in _async_schedule_add_entities_for_entry
    new_entities if type(new_entities) is list else list(new_entities)
                                                    ~~~~^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 56, in <genexpr>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, config_entry.options), d) for d in descriptions)
                       ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/entity.py", line 41, in create_entity
    entity = creator(description)
  File "/config/custom_components/solarman/sensor.py", line 56, in <lambda>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, config_entry.options), d) for d in descriptions)
                                               ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 44, in _create_entity
    return SolarmanNestedSensor(coordinator, description)
  File "/config/custom_components/solarman/sensor.py", line 99, in __init__
    self.coordinator.device.device_info[device_serial_number] = build_device_info(None, str(device_serial_number), None, None, None, parent_device_info["name"])
                                                                ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/common.py", line 106, in build_device_info
    device_info["identifiers"] = ({(DOMAIN, entry_id)} if entry_id else {}) | ({(DOMAIN, serial)} if serial else {})
                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for |: 'dict' and 'set'

### Version

25.05.08

### Home Assistant Core Version

2025.5.0

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-09:
Hi, @andreuka, thanks for letting me know, here is [fix: Device info build w/o entry_id](https://github.com/davidrapan/ha-solarman/commit/2f43914787d7afce91572f5720035ca910b26613)

closed 2025-05-09 by davidrapan

xarlien, 2025-05-09:
I'm sorry, maybe I'm misunderstanding. HA offered me an update for 25.05.08 (past this update NOT WORK DEYE entities). No other update to 25.05.08_01 is offered in HA. Should I manually update the file custom_components/solarman/common.py or will a new automatic update come out?

davidrapan, 2025-05-09:
If it's not offered then:
1. Open Solarman in HACS
2. Click Redownload under that 3 dots menu in upper right corner
3. Click DOWNLOAD

xarlien, 2025-05-12:
Thank for your help. How I can donate you? Paypal or something else?

davidrapan, 2025-05-12:
Hi @xarlien, you can using Github Sponsors by selecting the One-time tier if you want. 😉
[image]

xarlien, 2025-05-12:
Sorry, I try REVOLUT card but:

The credit card was declined by the payment processor, please verify the card's information. If the problem persists, please contact the card's issuer. You have not PayPal? I can send you 25USB via paypal.

davidrapan, 2025-05-12:
Yes I do have Paypal: my email visible on my profile here or `@DavidRapan`

xarlien, 2025-05-13:
Poslal jsem 500,- via paypal (až teď jsem si všimnul, že si čech) :-) Díky za tvou práci.

davidrapan, 2025-05-13:
Děkuju, moc si toho vážím. :)