davidrapan/ha-solarman #564: Deye 12K SG04LP3 parallel mode - entity is no longer being provided

000al000, 2025-05-09:
### Description

I've got three inverters Deye 12K SG04LP3 in paralell mode, use profile deye_p3 for all
After updating to last version a lot of sensors stop working on first (master) inverter

For example: DC Temperatur -entity is no longer being provided

Battery information also stop receiveing 
sensor.invertermaster01_battery - entity is no longer being provided

All sensros with Voltage and Power stop working

P.S. slave 2 inverters  working as early

### Reproduction Steps

three inverters Deye 12K SG04LP3 in paralell mode
profile deye_p3

### Log

2025-05-09 09:59:31.415 ERROR (MainThread) [custom_components.solarman.entity] Configuring {'name': 'Battery 1 Voltage', 'class': 'voltage', 'state_class': 'measurement', 'uom': 'V', 'scale': 0.1, 'rule': 1, 'registers': [10040], 'platform': 'sensor', 'key': 'battery_1_voltage_sensor', 'group': 'Battery 1', 'via_device': 'inverter', 'update_interval': 10, 'pack': 1, 'code': 3} failed. [TypeError: unsupported operand type(s) for |: 'dict' and 'set']
2025-05-09 09:59:31.417 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solarman platform for sensor: unsupported operand type(s) for |: 'dict' and 'set'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 390, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 681, in async_add_entities
    for entity in new_entities:
                  ^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 56, in <genexpr>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, config_entry.options), d) for d in descriptions)
                       ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/entity.py", line 41, in create_entity
    entity = creator(description)
  File "/config/custom_components/solarman/sensor.py", line 56, in <lambda>
    async_add_entities(create_entity(lambda x: _create_entity(coordinator, x, config_entry.options), d) for d in descriptions)
                                               ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/sensor.py", line 44, in _create_entity
    return SolarmanNestedSensor(coordinator, description)
  File "/config/custom_components/solarman/sensor.py", line 99, in __init__
    self.coordinator.device.device_info[device_serial_number] = build_device_info(None, str(device_serial_number), None, None, None, parent_device_info["name"])
                                                                ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/common.py", line 106, in build_device_info
    device_info["identifiers"] = ({(DOMAIN, entry_id)} if entry_id else {}) | ({(DOMAIN, serial)} if serial else {})
                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for |: 'dict' and 'set'
2025-05-09 09:59:31.488 WARNING (MainThread) [py.warnings] /usr/src/homeassistant/homeassistant/helpers/entity_registry.py:1149: RuntimeWarning: coroutine 'EntityPlatform._async_add_entity' was never awaited
  if value is not UNDEFINED and value != getattr(old, attr_name):

### Version

25.05.08

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

15.1

davidrapan, 2025-05-09:
Duplicate of #560, there's already new version 😉

davidrapan, 2025-05-09:
Duplicate of #560

closed 2025-05-09 by davidrapan