davidrapan/ha-solarman #568: Deye Battery BMS type support

rolupusoru, 2025-05-09:
### Is your feature request related to a problem?

Hi,

I have two Deye SE-G5.1 Pro-B batteries that communicate fine with Deye single phase 8Kw  inverter when ha-solarman is disabled. I can see each battery reporting its parameters to the cloud. However when ha-solarman is active i see BMS type is set to PYLON and after 5mins Solarman cloud will report both batteries are offline and stop reporting any data. It appears that ha-solarman interferes with inverter settings and changes battery BMS type.

### Describe the solution you'd like

Add a new "Unknown" option to BMS Battery Type, that will only read battery data from BMS if available and not send any BMS protocol configuration.

### Describe alternatives you've considered

None so far.

davidrapan, 2025-05-09:
Hi @rolupusoru, it's been discussed in #208

Lanzarhack, 2025-05-13:
I have the same problem.

rolupusoru, 2025-05-13:
I'm following instructions on #208, and changed `update_interval`  on` deye_hybrid.yaml` to 10seconds , but after HA restart i still have sensors update every 5 seconds. Might need to change on another file, any advice is welcomed.

davidrapan, 2025-05-13:
There are 3 occurrences of `update_interval: 5` in the file.

closed 2025-05-13 by rolupusoru