davidrapan/ha-solarman #569: Solarman failed to start with latest release.

heebo1974, 2025-05-10:
### Description

Updated to latest version today and solarman does not starts anymore

### Reproduction Steps

Updated solarman.

### Log

Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 08:47:18 (1 occurrences)
Last logged: 08:47:18
Error setting up entry 192.168.0.50 for solarman

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 56, in async_setup_entry
    await async_migrate_entries(hass, config_entry.entry_id, partial(migrate_unique_ids, config_entry))
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1720, in async_migrate_entries
    ent_reg.async_update_entity(entry.entity_id, **updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1234, in async_update_entity
    return self._async_update_entity(
           ~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<22 lines>...
        unit_of_measurement=unit_of_measurement,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 1172, in _async_update_entity
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Unique id '7600361584740ac322a07b4bb5f2cbfd_device_state_sensor' is already in use by 'sensor.solar_sofar_device_state'

### Version

25.05.08_1

### Home Assistant Core Version

2025.5.1

### Home Assistant Operating System Version

15.2

heebo1974, 2025-05-10:
Hmm.. It might be that part of the problem is my old inverter, which has been kept for historical reasons. It seems that during the Solarman update, the data from these inverters somehow got mixed up with each other?

eg. That old inverter model changed same as new one.
[image]

I disabled that old device, but it didn't help the issue and I don't know how to delete that device.

heebo1974, 2025-05-10:
I ended up to delete both devices and created new one again. That worked.
This case can be closed.

davidrapan, 2025-05-10:
Hi @heebo1974, because they had the same name (and proly serial number in the past too) therefore migration process most likely wasn't able to differentiate the two and was creating duplicate entries. Nothing I could prevent though even if I knew that something like this could happen.

Sorry for your troubles but I'm glad you sort it out! 😉

closed 2025-05-10 by davidrapan