davidrapan/ha-solarman #573: Switch Off Grid doesn't work in Deye 1 Phase

HarveyDent1, 2025-05-10:
### Description

Hello
```
      - name: Off Grid
        platform: switch
        rule: 1
        registers: [0x0119]
```
doesn't work in Deye SUN-6K-SG05LP1-EU

### Reproduction Steps

1. Go to Solarman
2. Press Off Grid Switch
Nothing is happening. Inverter sitll is On Grid

### Log

[home-assistant_solarman_2025-05-10T19-44-23.610Z.log](https://github.com/user-attachments/files/20141184/home-assistant_solarman_2025-05-10T19-44-23.610Z.log)

### Version

25.05.10

### Home Assistant Core Version

2025.4.4

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-10:
Hi @HarveyDent1, yeah it was a long shot and on top of that many configurables and whenever they work are heavily depended on firmware versions, etc. If you were looking for utilizing switch like this you could try something similar to #536 but with register address for single phase devices.

TonKro, 2025-06-08:
What does the Deye do actually in Off Grid mode practically?

closed 2025-07-23 by github-actions[bot]