davidrapan/ha-solarman #581: Huge spike in total production

omakoti, 2025-05-12:
What could be the cause of a huge spike in total production on solarpanels using solarman. This has happend a few times. The spike is around 40 GWh so it can't be some old history data.

[image]

davidrapan, 2025-05-12:
Hi @omakoti, what profile?

omakoti, 2025-05-13:
> Hi [@omakoti](https://github.com/omakoti), what profile?

Sorry, i don't know what do you mean by profile. 🙈

davidrapan, 2025-05-13:
Profile is selected during device configuration and is according to your inverter (brand/model).

omakoti, 2025-05-13:
> Profile is selected during device configuration and is according to your inverter (brand/model).

Can I see it somewhere afterwards?

omakoti, 2025-05-13:
> Yes, in device configuration.

sofar_lsw3

guermoule, 2025-05-13:
Hi
I have de same issue, for PV, Grid and load :

[image]

[image]

[image]

[image]

Profile : DEYE SG04LP3
Stick logger : Ethernet
Solarman version : 25.05.12

trutru21, 2025-05-13:
Hello! I've also encountered this weird spikes issue, and it was wrecking havoc in my utility meters logic so I had to create a filter for this. My profile is set to auto (I have Deye 10kw non-hybrid inverter).

davidrapan, 2025-05-13:
Can you guys enable debug logging, and if something like this happens again, write down the time, a few examples so I can find them in those logs.

guermoule, 2025-05-13:
Debug logging enbled, waiting case, but i see that !!

2025-05-13 09:21:48.022 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.11.9] Data: [1240, 5651, 86, 0, 64136, 63054, 195, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 4, 6, 19, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2298, 2300, 2299, 50, 0, 10, 145, 13, 41, 199, 199, 5000, 0, 145, 13, 41, 199, 2291, 2299, 2309, 0, 0, 0, 145, 13, 41, 199, 199, 5000, 0, 0, 0, 0, 0, 10, 7, 12, 0, 0, 0, 0, 0, 0, 0, 0, 910, 897, 0, 0, 2463, 36, 2429, 37, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2025-05-13 09:21:48.022 DEBUG (MainThread) [custom_components.solarman.device] [192.168.11.9] Querying 03 ❘ 0x03 ~ 0586 - 0708 ❘ 0x024A - 0x02C4 #123 succeeded
2025-05-13 09:21:48.023 DEBUG (MainThread) [custom_components.solarman.parser] Value: 15 of device_mppts_sensor validation failed: {'max': 12, 'default': 2}
2025-05-13 09:21:48.023 DEBUG (MainThread) [custom_components.solarman.parser] Value: 15 of device_phases_sensor validation failed: {'max': 3, 'default': 3}
2025-05-13 09:21:48.023 DEBUG (MainThread) [custom_components.solarman.parser] Value: 65535 of [61] is out of range: {'min': 0, 'max': 1000}

guermoule, 2025-05-13:
Hi
At 14:56:58 i have a spike (the browser is not i the same time zone than Inverter)

[image]

[image]

The Log 

[home-assistant_2025-05-13T15-23-09.599Z.log](https://github.com/user-attachments/files/20191141/home-assistant_2025-05-13T15-23-09.599Z.log)

guermoule, 2025-05-14:
Hi
An other case at 20:24:53

[home-assistant_solarman_2025-05-14T13-16-05.811Z.log](https://github.com/user-attachments/files/20208472/home-assistant_solarman_2025-05-14T13-16-05.811Z.log)

[image]

davidrapan, 2025-05-14:
In this case would be addition of some check very easy, that's really big number.. but those around 65k or 5k for PV Power not so much. :/

Add this to your Grid Power entity description:
```
        validation:
          min: -30000
          max: 30000
          invalidate_all:
```

guermoule, 2025-05-14:
ok for that number :-)

2025-05-14 18:38:53.204 ERROR (MainThread) [custom_components.solarman.parser] ParameterParser.try_parse: data: {(3, 62): [6405, 3602, 8711], (3, 210): [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], (3, 500): [2, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 0, 0, 0, 39, 25, 6485, 0, 5638, 0, 4, 0, 4491, 0, 3155, 0, 133, 19824, 0, 166, 0, 0, 0, 0, 23065, 0, 0, 0, 0, 0, 1250, 1455, 0, 0, 1, 0, 0, 0, 0, 1, 2048, 1, 5, 0, 0, 0, 0, 0, 0], (3, 586): [1270, 5407, 100, 0, 65471, 65278, 198, 0, 0, 0, 0, 0, 2343, 2343, 2345, 0, 0, 0, 0, 65533, 65527, 65524, 0, 5000, 100, 42, 47, 56, 123, 57, 65506, 176, 5, 151, 0, 0, 0, 65533, 65527, 65524, 0, 2346, 2346, 2338, 30, 0, 10, 142, 22, 54, 218, 218, 5000, 0, 142, 19, 45, 206, 2340, 2346, 2347, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}, definition: {'name': 'Grid Power', 'l': 1, 'description': 'Combined power of all three phases', 'class': 'power', 'state_class': 'measurement', 'uom': 'W', 'rule': 4, 'registers': [625, 690], 'icon': 'mdi:transmission-tower', 'attributes': ['inverse'], 'validation': {'min': -30000, 'max': 30000, 'invalidate_all': None}, 'platform': 'sensor', 'key': 'grid_power_sensor', 'group': 'Grid', 'update_interval': 15, 'code': 3} [ValueError: Invalidate complete dataset - 65524 of grid_power_sensor validation failed: {'min': -30000, 'max': 30000, 'invalidate_all': None}]

davidrapan, 2025-05-14:
I will try to implement some validation for energy sensors which would throw away sudden big increases in value. Just don't know how to define this big number yet. :)

omakoti, 2025-05-22:
Hi. It happened again. I thought the debug logging was on but it was not. Maybe the update couple of days a go switched it off so I don't have any logs to give. Looks like the spike is the same amount, about 40GWh.

[image]

xoferz, 2025-06-27:
Happens here as well, couple of times a day. Solarman integration 25.06.07

[image]

log:
`2025-06-27 11:37:40.881 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Scheduling 1 query request: [{'code': 3, 'start': 586, 'end': 708, 'count': 123}] #127840
2025-06-27 11:37:40.882 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Request 03 ❘ 0x03 ~ 0586 ❘ 0x024A: {'count': 123}
2025-06-27 11:37:40.882 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] SENT: f3 9a 00 00 00 06 01 03 02 4a 00 7b
2025-06-27 11:37:41.389 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] RECV: f3 9a 00 00 00 f9 01 03 f6 04 e2 00 00 00 00 00 00 00 00 ff ff 00 c8 00 00 ff ff 00 00 00 00 00 00 09 61 09 64 09 53 ff e7 ff fe 00 00 fb b3 fa 4c fc 7e f2 7d 00 00 13 88 01 fd 02 72 01 93 00 02 00 0c 00 04 00 01 00 01 00 01 00 03 00 00 ff fa fb b3 fa 4c fc 7e f2 7d 00 00 09 5f 09 5f 09 5d 02 6c 02 6c 02 6c 05 c8 05 c9 05 be 11 4f 11 4f 13 88 fc 18 01 7b 00 15 02 3c 03 cc 09 58 09 69 09 4c 00 00 00 00 00 00 01 7b 00 15 02 3c 03 cc 03 cc 13 88 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 0b 00 c4 00 00 00 00 14 3e 00 34 0d c4 00 37 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 00
2025-06-27 11:37:41.390 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] DATA: [1250, 0, 0, 0, 0, 65535, 200, 0, 65535, 0, 0, 0, 2401, 2404, 2387, 65511, 65534, 0, 64435, 64076, 64638, 62077, 0, 5000, 509, 626, 403, 2, 12, 4, 1, 1, 1, 3, 0, 65530, 64435, 64076, 64638, 62077, 0, 2399, 2399, 2397, 620, 620, 620, 1480, 1481, 1470, 4431, 4431, 5000, 64536, 379, 21, 572, 972, 2392, 2409, 2380, 0, 0, 0, 379, 21, 572, 972, 972, 5000, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 267, 196, 0, 0, 5182, 52, 3524, 55, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]
2025-06-27 11:37:41.391 DEBUG (MainThread) [custom_components.solarman.parser] grid_power_factor_number: 65530 is outside of range: {'min': 0, 'max': 1000}
2025-06-27 11:37:41.392 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Returning 67 new values
2025-06-27 11:37:41.393 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.514 seconds (success: True)
2025-06-27 11:37:45.881 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Scheduling 1 query request: [{'code': 3, 'start': 586, 'end': 708, 'count': 123}] #127845
2025-06-27 11:37:45.881 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Request 03 ❘ 0x03 ~ 0586 ❘ 0x024A: {'count': 123}
2025-06-27 11:37:45.882 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] SENT: 4e 02 00 00 00 06 01 03 02 4a 00 7b
2025-06-27 11:37:46.092 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] RECV: 4e 02 00 00 00 f9 01 03 f6 04 e2 00 00 00 00 00 00 00 00 ff ff 00 c8 00 00 ff ff 00 00 00 00 00 00 09 63 09 66 09 56 ff e4 ff fe 00 00 fb b1 fa 4b fc 7e f2 7a 00 00 13 8c 01 fd 02 72 01 93 00 01 00 0c 00 04 00 01 00 01 00 01 00 03 00 00 ff fa fb b1 fa 4b fc 7e f2 7a 00 00 09 65 09 63 09 62 02 6c 02 6c 02 6c 05 c9 05 ca 05 c0 11 53 11 53 13 8c fc 18 01 7a 00 15 02 3e 03 cd 09 5b 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2025-06-27 11:37:46.093 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] DATA: [1250, 0, 0, 0, 0, 65535, 200, 0, 65535, 0, 0, 0, 2403, 2406, 2390, 65508, 65534, 0, 64433, 64075, 64638, 62074, 0, 5004, 509, 626, 403, 1, 12, 4, 1, 1, 1, 3, 0, 65530, 64433, 64075, 64638, 62074, 0, 2405, 2403, 2402, 620, 620, 620, 1481, 1482, 1472, 4435, 4435, 5004, 64536, 378, 21, 574, 973, 2395, 2304, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2025-06-27 11:37:46.095 DEBUG (MainThread) [custom_components.solarman.parser] grid_power_factor_number: 65530 is outside of range: {'min': 0, 'max': 1000}
2025-06-27 11:37:46.096 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Returning 67 new values
2025-06-27 11:37:46.097 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.218 seconds (success: True)
2025-06-27 11:37:50.882 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Scheduling 1 query request: [{'code': 3, 'start': 586, 'end': 708, 'count': 123}] #127850
2025-06-27 11:37:50.883 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Request 03 ❘ 0x03 ~ 0586 ❘ 0x024A: {'count': 123}
2025-06-27 11:37:50.883 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [10.20.110.55] SENT: 9d bc 00 00 00 06 01 03 02 4a 00 7b
[... cut ...] 5535, 0, 0, 0, 0, 0, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 0, 0, 0, 0, 0]
2025-06-27 11:37:51.333 DEBUG (MainThread) [custom_components.solarman.parser] grid_power_factor_number: 65530 is outside of range: {'min': 0, 'max': 1000}
2025-06-27 11:37:51.335 DEBUG (MainThread) [custom_components.solarman.device] [10.20.110.55] Returning 67 new values
2025-06-27 11:37:51.335 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.457 seconds (success: True)
`

davidrapan, 2025-06-27:
Corruption of power sensors is hard to detect as these values could as well be valid for some devices... but detection of corrupted energy data is already in place.

davidrapan, 2025-06-27:
> Sorry if this is a different issue

No really, but issue is originating in the device and integration only does some mitigation throughout value validation, but this will never work 100%.

> Could we set a limit? There is no way my 25A*220v circuit would provide 64kW.

Yours maybe not, but there are devices well beyond 100kW.

> These spikes translate to corrupted energy dashboard for me.

Well, why you don't show which energy sensor causing the issue instead of power ones?

xoferz, 2025-06-28:
> No really, but issue is originating in the device and integration only does some mitigation throughout value validation, but this will never work 100%.
>
>
> Yours maybe not, but there are devices well [...]

I meant more like a user configurable limit. If power reported is above X (ca 60kW for me), discard the reading (or equate it to previous reading). For me the wrong spikes seem always to be around 64k 

[image]

> 
> > These spikes translate to corrupted energy dashboard for me.
> 
> Well, why you don't show which energy sensor is causing the issue instead of power ones?

Now that I look at it, the integration does not offer grid used energy per phase, it is an integral sensor based on phase power reading I have created. If it is not feasible to filter out obviously wrong power readings at the integration level, I guess I can discard them myself. I just chimed in as it seems to be the same problem.

davidrapan, 2025-06-30:
Log?

sskom, 2025-07-01:
Here is one more spike

This error originated from a custom integration.

Logger: custom_components.solarman.coordinator
Source: helpers/update_coordinator.py:414
integration: Solarman (documentation, issues)
First occurred: June 30, 2025 at 18:29:24 (148 occurrences)
Last logged: 11:22:16

Error fetching Deye2 data: Invalidate complete dataset. total_production_sensor: 15361.900000000001 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}
Error fetching Deye1 data: Invalidate complete dataset. total_production_sensor: 16954.3 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}
Error fetching Deye1 data: Invalidate complete dataset. total_production_sensor: 16954.9 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}
Error fetching Deye1 data: Invalidate complete dataset. total_production_sensor: 16955.4 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}
Error fetching Deye2 data: Invalidate complete dataset. total_production_sensor: 15363.1 validation failed. Conditions: {'min': 0.1, 'dev': 100, 'invalidate_all': 2}

[image]

sskom, 2025-07-01:
> You need to download the log file where the error entries are with date and time.  😉

Sorry for previous unful information. I attached file with the log with 5 minutes (from 11:20 till 11:25). The spike was registered at 11:22

[home-assistant_2025-07-01T11-16-59.736Z.log](https://github.com/user-attachments/files/20997586/home-assistant_2025-07-01T11-16-59.736Z.log)

davidrapan, 2025-07-01:
These are today's sensors. Are you experiencing this w/ total's too?

davidrapan, 2025-07-01:
This is what I'm interested in. I want a debug log of this fluke. Today's are not really important.

davidrapan, 2025-07-01:
You will have to manually download the latest source code and upload it to your HA.

davidrapan, 2025-07-01:
This value is "estimated" within integration so it's not always 100%.. Look into sensor attributes. There are the states from which it's calculated.

You can try to clear this attribute using Developer tools > States and then restart HA to restart the process.

davidrapan, 2025-07-01:
Yeah, it might refresh the states faster than it can restart... but that one big number is skewing it (not even that particular number can be changed/removed?). I'm not sure, but it's most likely related to data corruption. 😉

The second option is to remove the device and re-add it back again. 😆

sskom, 2025-07-28:
Hi David! Here is the spike in Total Production and the piece of log from 3 am to 4 pm. Hope it will help. Thank you!

[home-assistant_2025-07-28T04-41-30.306Z.log](https://github.com/user-attachments/files/21460984/home-assistant_2025-07-28T04-41-30.306Z.log)

[image]

norman-rennhack, 2025-08-04:
I do also have some spikes in my daily production from time to time, which are not shown in solarman itself. Mostly in the morning hours but lately also observed in the evening.

[image]
[image]
[image]
[image]

davidrapan, 2025-08-14:
What is the state of the matter in the latest release?

norman-rennhack, 2025-12-14:
Still having wrong spikes (even though much smaller ones) e.g. today morning at 09:30 am the integration shows 200wh but there wasn’t anything produced at this time.

[image]
[image]
[image]

davidrapan, 2025-12-15:
Unfortunately, there is nothing we can do about such small discrepancies.