davidrapan/ha-solarman #583: Suggestion: Add "Sell from Battery" Option in Work Mode

Zhyvotovskyy, 2025-05-13:
### Is your feature request related to a problem?

Hi David,

I have a suggestion to improve the “System Work Mode” configuration screen on the inverter.

Currently, in Sell mode, there is no way to control whether the battery is allowed to discharge into the grid. It would be very helpful to add an additional checkbox option like:

✅ Sell from Battery

This would allow users to precisely control when battery energy can be exported — for example, when operating under a Feed-in Tariff (Green Tariff) program.
If the box is checked, the battery is allowed to export energy to the grid. If unchecked, only PV surplus would be exported.

This would help reduce unnecessary battery cycling and give users better control over energy flow.

I’m happy to help with ideas or testing if needed.

Best regards,
Mykola

[image]

### Describe the solution you'd like

Add an additional checkbox for "Sell from Battery" in the "System Work Mode" configuration.  
This would allow users to explicitly control whether the battery is allowed to discharge into the grid during Sell mode.

**Use case:**  
For users with a Feed-in Tariff (Green Tariff), it's useful to export from the battery only during specific hours to increase earnings and avoid unnecessary cycling of the battery.

If checked — the battery is allowed to export energy.  
If unchecked — only PV surplus will be sold to the grid.

### Describe alternatives you've considered

Currently, there is no direct way to prevent or allow battery export in Sell mode based on time.  
Some users are using indirect workarounds through time-of-use charging settings, but they are not reliable or intuitive.

An explicit checkbox would greatly improve control and clarity for end users.

rolupusoru, 2025-05-13:
To  my knowledge if you want to discharge battery  into grid, then you need to set your inverter on "Seling first" under System Work Mode first page. Then on time of use you set the interval and battery % you want your the inverter to discharge. 

In your example, with this setup, you inverter would discharge the battery down to 55% between 6-11 in the morning and again between 18-23. the rest of the time it will look for power to charge you batteries back to 100%. Because you also have grid charge set, it will also take power from grid to charge them. If you uncheck grid charge, then inverter will only charge from solar if available.

Zhyvotovskyy, 2025-05-14:
Hi David,

Thank you for your detailed explanation — it helped clarify the inverter
logic.

I’d like to kindly suggest a small but very useful addition to your
integration. In my setup, I often need to *dynamically enable or disable
battery discharge to the grid*, depending on the current situation.

For example, during an *air raid alert in Ukraine*, I *disable battery
export* to preserve charge in case of a sudden blackout. When the situation
is calm ("clear skies" — both literally and figuratively), I allow the
battery to discharge to the grid again. Being able to control this through
Home Assistant would make this much easier and safer.

There’s also another reason. My *inverter is 50 kW*, but I’m only allowed
to *export 30 kW*. That means I need to intelligently *balance* battery
usage: charge the battery during solar peak, and discharge it later when
solar output drops. In summer and winter, my logic is different — so *automated
control is essential*.

At the moment, I can manage everything from Home Assistant — except for
toggling that critical checkbox to allow battery discharge to the grid. I
still have to do that manually through the mobile app.

Would it be possible to expose that option as a switch or setting in the
integration? I’d be happy to assist with testing or documentation if needed.

Thank you again for your great work!

Best regards,
Mykola from Ukraine

Adminius, 2025-05-14:
> [image]

Which inverter and firmware do you have?

andriiko78, 2025-05-23:
Hello.
My inverter Deye SUN-50K also has these menu items.
Firmware version MAIN：3103-1095-1E08

andriiko78, 2025-05-23:
> Привіт, Девіде, дякую за твою роботу над інтеграцією ha-solarman. Просто хотів повідомити, що після оновлення до версії *v25.05.20*, інтеграція втратила зв'язок з інвертором. Ми повернулися до *v25.05.12*, і все знову працює добре. Будь ласка, повідом мене, якщо тобі потрібні якісь логи або допомога в тестуванні. З найкращими побажаннями, Микола, пт, 23 травня 2025 р. о 11:36, andriiko78 ***@***.***>:
> […](#)

I have a similar problem. The latest version of the program does not read data from the inverter.

davidrapan, 2025-05-23:
Can someone provide debug log from latest version?

closed 2025-07-20 by github-actions[bot]