davidrapan/ha-solarman #586: Connection is reset by the peer. Will try to restart the connectiom - ConnectionRefusedError

dar3khudy, 2025-05-14:
### Description

Inverter Sofar - 6.6KTL-x with lse-3 card (SF4ES006) connected by LAN (card should work as lsw-3 as they are equal)
Everything is connecting but on the middle of the day it is getting disconnected - I can ping datamanager but HA can't login into and after some time it is back in operation.
Because of that the active power looks like:
[image]
I'm not sure but it seems to me that the problem lies in the frequency of inverter polling, is it possible to set the polling interval synchronously e.g. every 5 or 10s rather than async?

[update interval.csv](https://github.com/user-attachments/files/20209177/update.interval.csv)

### Reproduction Steps

In my case, this is how it has worked from the beginning - I didn't have to do anything to get it.

### Log

[log.txt](https://github.com/user-attachments/files/20208895/log.txt)

### Version

25.05.12

### Home Assistant Core Version

2025.5.1

### Home Assistant Operating System Version

15.2

dar3khudy, 2025-05-15:
Yes I do have, for a test today I setup this extension like:
TIMINGS_INTERVAL = 15
TIMINGS_INTERVAL_SCALE = 2
in const.py

Like expected data is coming every ~30s, but till now I'm not getting disconnected (yesterday I got first disconnection about this hour).

dar3khudy, 2025-05-16:
Ok I revert the change for TIMINGS_INTERVAL, and apply only TIMINGS_INTERVAL_SCALE = 6.
But it still has a problem with disconnections:
2025-05-16 08:17:42.772 ERROR (MainThread) [custom_components.solarman.coordinator] Error fetching Inverter data: [192.168.0.30] Cannot open connection
And after couple of minutes it is coming back, but all the time I can access the web interface of my inverter (so datamanager card is working), with  TIMINGS_INTERVAL_SCALE = 6 it is happening less often but it is still there what could be a reason of it?

dar3khudy, 2025-05-16:
ok done, I will tests and let you know, no idea why it is disconnecting and not let connect back?
On the config_hide.html there is something like:
[image]
Perhaps that timeout I need to extend also?

davidrapan, 2025-05-16:
Mine is set to 300.

dar3khudy, 2025-05-17:
I also set it for 300, but nothing changes, maybe it loses connection less often, or maybe it's a coincidence, but it still happens and I don't know the cause.

davidrapan, 2025-05-18:
I honestly have no idea. It's strange that the logger refuses to communicate for such a long time windows. Did you do the test pings directly from HA instance? To rule out possibility that it's a network issue.

dar3khudy, 2025-05-19:
It seems that when it throws away the HA connection, it doesn't respond to the ping either, but first disconnects the solarman and then doesn't respond to the ping, and when it returns, it reconnects to the solarman:
binary_sensor.192_168_0_30,on,2025-05-19T04:00:00.000Z
binary_sensor.192_168_0_30,off,2025-05-19T08:14:13.661Z
binary_sensor.192_168_0_30,on,2025-05-19T08:15:26.686Z
binary_sensor.192_168_0_30,off,2025-05-19T10:13:57.657Z
binary_sensor.192_168_0_30,on,2025-05-19T10:14:31.713Z
binary_sensor.192_168_0_30,off,2025-05-19T12:13:57.659Z
binary_sensor.192_168_0_30,on,2025-05-19T12:14:31.687Z

sensor.inverter_output_active_power,430,2025-05-19T08:01:52.927Z
sensor.inverter_output_active_power,unavailable,2025-05-19T08:04:02.653Z
sensor.inverter_output_active_power,540,2025-05-19T08:15:22.915Z
sensor.inverter_output_active_power,820,2025-05-19T10:03:44.971Z
sensor.inverter_output_active_power,unavailable,2025-05-19T10:05:54.657Z
sensor.inverter_output_active_power,710,2025-05-19T10:14:26.145Z
sensor.inverter_output_active_power,740,2025-05-19T10:15:15.910Z
sensor.inverter_output_active_power,880,2025-05-19T11:17:46.963Z
sensor.inverter_output_active_power,unavailable,2025-05-19T11:19:56.653Z
sensor.inverter_output_active_power,800,2025-05-19T11:51:50.203Z
sensor.inverter_output_active_power,820,2025-05-19T11:52:46.853Z

So it seems that for unknown reasons the data manager card jammed and probably there was a reboot of the watchdog and during it the network faded for a while and then it worked again, so in my opinion the network fading is a symptom of the problem with this communication, not the cause - this one remains unknown (yesterday I removed the solarman integration from the HA and did not lose a single ping between the HA and the inverter).

Excerpt from log, before and after disconnection around 10 o'clock:

2025-05-19 10:01:02.611 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Scheduling 1 query request. ^18000
2025-05-19 10:01:02.611 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Querying 03 ❘ 0x03 ~ 0000 - 0039 ❘ 0x0000 - 0x0027 #040 ...
2025-05-19 10:01:02.612 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] SENT: a5 17 00 10 45 b6 00 c4 6b 5d 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 28 45 d4 72 15
2025-05-19 10:01:03.002 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] RECD: a5 63 00 10 15 b6 96 c4 6b 5d 7d 02 01 87 96 ee 03 f4 45 00 00 b7 4e 3c 64 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 0e 2d 00 3c 0c 88 00 3e 00 15 00 13 00 28 ff c5 13 83 08 d4 00 6a 08 d1 00 6a 08 f5 00 68 00 00 6f 13 00 00 45 81 00 64 01 16 00 16 00 25 18 60 0e 13 0c 6d 00 3c 00 00 00 01 00 00 04 15 08 51 06 f9 00 0c 64 d2 03 15
2025-05-19 10:01:03.003 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] Data: [2, 0, 0, 0, 0, 0, 3629, 60, 3208, 62, 21, 19, 40, 65477, 4995, 2260, 106, 2257, 106, 2293, 104, 0, 28435, 0, 17793, 100, 278, 22, 37, 6240, 3603, 3181, 60, 0, 1, 0, 1045, 2129, 1785, 12]
2025-05-19 10:01:03.003 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Querying 03 ❘ 0x03 ~ 0000 - 0039 ❘ 0x0000 - 0x0027 #040 succeeded
2025-05-19 10:01:03.003 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Returning 39 new values. [Previous State: Connected (1)]
2025-05-19 10:01:03.003 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.392 seconds (success: True)
2025-05-19 10:01:52.611 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Scheduling 1 query request. ^18050
2025-05-19 10:01:52.611 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Querying 03 ❘ 0x03 ~ 0000 - 0039 ❘ 0x0000 - 0x0027 #040 ...
2025-05-19 10:01:52.611 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] SENT: a5 17 00 10 45 b7 00 c4 6b 5d 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 28 45 d4 73 15
2025-05-19 10:01:52.926 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] RECD: a5 63 00 10 15 b7 97 c4 6b 5d 7d 02 01 b8 96 ee 03 26 46 00 00 b7 4e 3c 64 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 0e 31 00 40 0c d5 00 41 00 17 00 15 00 2b ff c8 13 82 08 cc 00 6a 08 bd 00 6d 08 f0 00 69 00 00 6f 13 00 00 45 81 00 65 01 17 00 16 00 26 18 5f 0e 17 0c bd 00 3c 00 00 00 01 00 00 04 15 08 51 06 f9 00 0c a3 e8 58 15
2025-05-19 10:01:52.926 DEBUG (MainThread) [custom_components.solarman.pysolarman.pysolarman] [192.168.0.30] Data: [2, 0, 0, 0, 0, 0, 3633, 64, 3285, 65, 23, 21, 43, 65480, 4994, 2252, 106, 2237, 109, 2288, 105, 0, 28435, 0, 17793, 101, 279, 22, 38, 6239, 3607, 3261, 60, 0, 1, 0, 1045, 2129, 1785, 12]
2025-05-19 10:01:52.926 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Querying 03 ❘ 0x03 ~ 0000 - 0039 ❘ 0x0000 - 0x0027 #040 succeeded
2025-05-19 10:01:52.927 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Returning 39 new values. [Previous State: Connected (1)]
2025-05-19 10:01:52.927 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.316 seconds (success: True)
2025-05-19 10:02:42.611 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Scheduling 1 query request. ^18100
[... cut ...] 3365, 60, 0, 1, 0, 1045, 2129, 1785, 12]
2025-05-19 12:22:46.997 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Querying 03 ❘ 0x03 ~ 0000 - 0039 ❘ 0x0000 - 0x0027 #040 succeeded
2025-05-19 12:22:46.997 DEBUG (MainThread) [custom_components.solarman.device] [192.168.0.30] Returning 39 new values. [Previous State: Connected (1)]
2025-05-19 12:22:46.997 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 0.385 seconds (success: True)

davidrapan, 2025-05-25:
I mean, there's not that much data being read...

closed 2025-07-10 by github-actions[bot]