davidrapan/ha-solarman #588: Deye SUN-20K-SG05LP3-EU-SM2, max grid/pv charging limited to approx. 12kw

robertsze, 2025-05-15:
### Description

The new Deye 20K inverter works perfectly with this integration, but for some reason it will never load the battery with more than 12kw. As this integration was tested with the 12K inverter only, maybe some register is limiting it to 12kw. (A few other registers had to be adjusted too already, but cannot find anything related to charge limiting)

### Reproduction Steps

Just enable grid charging an see max. charging is limited to approx. 12kw

### Log

There is no log available

### Version

25.05.12

### Home Assistant Core Version

2025.5.1

### Home Assistant Operating System Version

15.2

robertsze, 2025-05-15:
Max loading current and max battery loading current has been adjusted to 350A already, so this is not the limiting factor.

davidrapan, 2025-05-15:
[fix: Increased max power and current range in Deye 3P](https://github.com/davidrapan/ha-solarman/commit/f349f9dff9e10508c42107c1f8e641c55e46dd6e)

closed 2025-05-15 by davidrapan

robertsze, 2025-05-15:
I previously tried all the values from this fix already (and also applied this changes again), but unfortunately charging is still limited to approx. 12kw only. Maybe there is an additional register to write for the 20K version to allow charging with more than 12kw...

davidrapan, 2025-05-15:
Are you sure your batteries allow this? What you have in Battery sensor attributes?

robertsze, 2025-05-15:
yes, they do, was also on a call with electrician/deye support and they confirmed they should charge with up to 350A

Nominal Voltage  48
Life Cycle Rating 6.000
Corrected Capacity 200
BMS SOC 81
BMS Current -233
BMS Voltage 54,1
BMS Other Symbol 0
BMS Charging Voltage 58,4
BMS Discharging Voltage 48
BMS Charging Current 480
BMS Discharging Current 1.000
BMS Max Charging Current 500
BMS Max Discharging Current 1.020

deye_battery_power is always around/between -12.200 and -13.064 while charging from grid

davidrapan, 2025-05-15:
Hm, interesting.. and have you checked if the charging values ​​are set correctly directly on the device?

robertsze, 2025-05-15:
yes, all "visible" values on the Deye itself look correct, even support said everything looks correct and they assume that maybe "invisible registers / not visible values on the display" have been written to with "wrong" values, but I somewhat doubt this, at least all other registers from the configuration file and their values don't look really suspicious, at least to me

davidrapan, 2025-05-15:
There's no such thing as invisible "registers"... on top of that integration only configures what you can see in particular profile. And it's pretty much the same what you can control directly using the display.

This max charge current MUST be controllable from the device, if it is not the case, something is wrong and you should urge your support to fix it. What a lazy as* nonsense response from them... jeez

robertsze, 2025-05-16:
The max charge current is controllable from the device and is set correctly, we are still checking if writing "12kw max settings" "misconfigured" something initially

davidrapan, 2025-05-16:
> Can you think of any register/value (which is defined in the configuration) which could maybe limit it? (which is normally not set in the inverter, but only when the integration gets activated)

Nothing like this is happening...  you control the inverter, through integration. Integration doesn't control anything on it's own.