davidrapan/ha-solarman #590: max_size improvements?

voed, 2025-05-15:
### Is your feature request related to a problem?

Hi. I am using Modbus-TCP protocol with deye_hybrid. As we figured out in #310, it requires to set "max_size" to make it work. So it's a bit annoying to apply that change after every update. Maybe we improve it somehow?

### Describe the solution you'd like

I am not sure, maybe it's related to my usb-modbus controller, or just [modbus-tcp gateway](https://github.com/kellerza/sunsynk/tree/main/hass-addon-mbusd), but maybe we can make it select optimal max size automatically?

### Describe alternatives you've considered

Or maybe just move max_size option to the integration's GUI, so updates will not override it:)

davidrapan, 2025-05-15:
Hi @voed, `max_size: 20` is too small to even consider adding it as the default. On top of that you need to use d`eye_hybrid.yaml` profile copied over to `custom/` folder anyway, no?

voed, 2025-05-16:
Somehow i am missed fact that i need to use `custom` folder for 3rd party profiles. I was just using `inverter_definitions` folder, which is overrides on updates.
I guess it's fine than. Thanks.

davidrapan, 2025-05-16:
No worries. 😉

closed 2025-05-16 by davidrapan