davidrapan/ha-solarman #592: Deye Inverter Today's Losses resets mid-day

plebann, 2025-05-16:
### Description

[image]

I think it's like that since 12 of March.

[image]

### Reproduction Steps

Sensor sensor.inverter_today_losses resets (and stays at 0) at some point in day (around 1,2-1,5 kWh).
Hard to say what time, but it's rather early, this week statistics

[image]

[image]

[image]

[image]

[image]

### Log

No logs at this time

### Version

25.05.12

### Home Assistant Core Version

2025.5.1

### Home Assistant Operating System Version

15.2

davidrapan, 2025-05-16:
Hi @plebann, I will need the debug log + graphs of all today's sensors for the specific day this happens. 😉

davidrapan, 2025-05-16:
Yes.

plebann, 2025-05-17:
[home-assistant_solarman_2025-05-17T09-25-41.529Z.zip](https://github.com/user-attachments/files/20264000/home-assistant_solarman_2025-05-17T09-25-41.529Z.zip)

davidrapan, 2025-05-18:
And graphs from that day? (of all **Total** sensors)

plebann, 2025-05-18:
[image]

You wrote "total sensors", what else should I attached?

davidrapan, 2025-05-18:
Ah, sorry, I meant **Today** sensors: Today Production, Today Energy Import, Today Energy Export, Today Battery Charge, Today Battery Discharge and Today Load Consumption.

plebann, 2025-05-18:
Not sure if this is visible enough:

[image]

plebann, 2025-05-18:
[image]

davidrapan, 2025-05-18:
And the time when it happened, exactly?

BTW, why it reads all Battery X sensors when you proly don't have any?

plebann, 2025-05-18:
[image]
Not sure what do you mean by this battery sensors?
I have one battery, hight voltage one.

davidrapan, 2025-05-18:
> And the time when it happened, exactly?

I mean, write the time, please.

> Not sure what do you mean by this battery sensors?

I can see in the log that you have added Battery 1 - 12 sensors. They should be added as separate devices connected through the inverter. What value is Number of Battery packs in device configuration > Additional options set to?

plebann, 2025-05-18:
> What value is Number of Battery packs in device configuration > Additional options set to?
Looks like "-1". I'll change that.

[image]

davidrapan, 2025-05-18:
Hm, I have honestly no idea what's going on in there... But I can see that even Total Losses sensor validation fails cause it's 0. Can you show me current values of Total sensors?

plebann, 2025-05-18:
From what I can see it's showing Unknown, and no history available.
Last time it was available was 9 of May:

[image]

But last meaningful values was 11 of March:

[image]

davidrapan, 2025-05-18:
No, I wanted to see just values of ALL Total * sensors, from device detail page.

plebann, 2025-05-18:
I hope this is it:

[image]

davidrapan, 2025-05-18:
Yes it is.

What device this is? Single phase? What's going on w/ that Today's Generation? Do you have any idea? You proly created Utility Meter on top of it w/ a reset, am I right?

plebann, 2025-05-18:
SG01HP3
I created this meter some time ago as there was some weird issue with this sensor not being reset on a daily basis.
But recently it's working fine, so I guess I should delete it.

plebann, 2025-05-18:
>  because Production sensor was already tracking Generation.

That's now that I see compared to solarman data:

[image]

[image]

davidrapan, 2025-05-18:
Show me also values of Today sensors, please.

plebann, 2025-05-18:
[image]

davidrapan, 2025-05-18:
Hmmm, okay so your device needs to calculate the losses w/ Gen port input...

Can you show me an output of Developer tools > ACTIONS > Solarman: Read Holding Registers (Modbus Function Code 3), select your device, set address to 500 and count to 50.

plebann, 2025-05-18:
`"500": 2
"501": 0
"502": 0
"503": 0
"504": 0
"505": 0
"506": 0
"507": 0
"508": 46134
"509": 39092
"510": 0
"511": 0
"512": 0
"513": 0
"514": 164
"515": 36
"516": 11504
"517": 0
"518": 9530
"519": 0
"520": 0
"521": 235
"522": 204
"523": 0
"524": 32016
"525": 0
"526": 16
"527": 886
"528": 0
"529": 357
"530": 0
"531": 0
"532": 0
"533": 0
"534": 32617
"535": 0
"536": 62
"537": 5314
"538": 0
"539": 0
"540": 1250
"541": 1238
"542": 0
"543": 0
"544": 1
"545": 0
"546": 0
"547": 0
"548": 0
"549": 0
`

davidrapan, 2025-05-18:
I'm gonna have to figure out how how to add possibility to switch losses calculation w/ or w/o gen port.

In the meantime you can add addition of gen port to Today Losses:

plebann, 2025-05-18:
This works, thanks

plebann, 2025-05-18:
Maybe the difference of Production sensor already tracking Generation is a matter of HP/LP versions of Deye inverter? Just a thought.

davidrapan, 2025-05-18:
That was my first thought too but I bet it's about SG01 vs SG04 or SG05 because former would be too easy (it's detectable). On top of that I have no idea how those register work when switched to SmartLoad...

davidrapan, 2025-05-18:
In #435 is why it was removed.

davidrapan, 2025-05-19:
So I decided to solve it for now based on IO Mode selection in [feat: Conditional name and use of generator energy sensors](https://github.com/davidrapan/ha-solarman/commit/e82fd3280b67197903e366e472d62fe4ce94fdb3)

closed 2025-05-21 by davidrapan

davidrapan, 2025-05-21:
Hey @plebann, if you could report on the latest update, if everything works as expected would be highly appreciated. 😉

plebann, 2025-05-21:
I'm not sure yet, but it seems it's back to previus, Today's losses become 0 after update:

[image]

Inverter Total Losses Unknown

davidrapan, 2025-05-21:
Thanks! It turns out there was a small error, so here it is [fix: Conditional use of generator port in calculation of losses](https://github.com/davidrapan/ha-solarman/commit/9554ef33bca9302809fdce85632c8c0b5c51ed62)

plebann, 2025-05-22:
After applying that patch, it's working.

davidrapan, 2025-05-22:
Awesome! 🎉